EdgeTX/edgetx #2824: Emergency Mode after editing a theme and saving the changes

txbone66, 2022-12-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have 2 Radiomaster TX16S controllers.  One is an 'original' and one is a MKII.  I upgraded both to 2.8 and saw that when I make an edit to a theme and save it, the MKII goes into Emergency Mode while the original TX16S does just fine.  The SD Card contents are the same for both radios.

### Expected Behavior

The radio should return to the theme screen and let you continue working.  Right now I have to power cycle the radio to see the changes applied on the MK II

### Steps To Reproduce

Set a theme to active.
Edit the theme, just change any color.
When I go to save the change, the MK II goes into emergency mode.

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

gagarinlg, 2022-12-10:
I do not think that this depends on the radio hardware. Can you swap cards between the radios and try again?

txbone66, 2022-12-10:
So I did another test.  I put my backup in my old radio that was on 2.7.1, flashed the bootloader, and then flashed the firmware to update to 2.8.0. No issues and then went to edit a theme, did that, saved it and did not enter emergency mode.  Repeated on the MKII and now I'm not getting the issue on EM with a theme change.  No idea why it's different today.

closed 2022-12-10 by txbone66

gagarinlg, 2022-12-11:
Thx for testing.