EdgeTX/edgetx #2825: Mac version of Companion crashes with model file open when minimized.

jmxp69, 2022-12-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Companion 2.8
MacOS 13.0.1

Minimize Companion with an ETX document open and it crashes. Easily repeatable. Does not matter if the ETX file has models or not. Does not matter if the ETX file has been saved or not.

### Expected Behavior

Companion shouldn't crash when minimized with an open ETX file.

### Steps To Reproduce

Open Companion 2.8
Open or create a model file
Minimize Companion
See Crash

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

[image]
[image]

raphaelcoeffic, 2022-12-10:
@jmxp69 we might have a rather painful journey ahead.... ATM, we are using Qt 5.12.x, which is rather old and does not support explicitly newer MacOS versions or even M1 Macs... However, moving to 6.x is going to be *very* painful, so that we might attempt to move to latest 5.x first (5.15.x), which will have native support for M1 Macs on open source version in March.

philmoz, 2022-12-12:
Companion and simulator build and seem to run ok with Qt 5.15.2 on MacOS.
Only tested on MacOS 12.6.1 (Intel) at this stage.

jmxp69, 2022-12-12:
My machine is a 2022 M2 FWIW.

raphaelcoeffic, 2022-12-12:
Works ok for me as well on MBP M1 Pro with macOS 12.6.1.

pfeerick, 2022-12-12:
It crashed on minimise when I tried it on macOS 13 / Ventura using an emulator - I did try to save the log but haven't logged into that machine since to add it to the issue.

raphaelcoeffic, 2022-12-12:
Ok, so it seems to be related to macOS 13.... The tough part is that Qt 5.12 does not support it officially, nor does any open source 5.x version. We might really have to move to Qt 6 rather sooner than later (or find some kind of work-around).

*Side note*: on my side, when compiling with latest macOS SDK, Companion gets stuck at the splash screen. If I remove the splash screen, everything works as expected.

elecpower, 2022-12-12:
Qt 6 drops support for Win 32 bit refer https://doc.qt.io/qt-6/supported-platforms.html. Some features we use depreciated at 5.15 and removed in 6 though most replaced albeit with changed methods.
It is not unreasonable to drop 32 bit as this has been occurring with many products. However our workload to refactor and test will be significant especially since we have to get working across multiple OSes.
I haven't researched this yet but another challenge could be Qt licensing and/or accessing the same version across all toolchains. Qt changed the rules for 5.15.

elecpower, 2022-12-12:
If we decide to bit the bullet on Qt then IMO we should look at updating the versions of SDL, etc

stronnag, 2022-12-12:
another FYI; companion / sim builds and runs fine with 5.15 on Arch Linux (gcc 12) and (not even supported) FreeBSD 13.1, clang 13). Deprecations not withstanding.

pfeerick, 2022-12-13:
For what it's worth, this was the log that Apple spat at me on crash... 
<details>
<summary>Log</summary>

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               companion [560]
Path:                  /Applications/EdgeTX Companion 2.8.app/Contents/MacOS/companion
Identifier:            org.edgetx.companion
Version:                ()
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-12-13 10:39:55.6242 +1000
OS Version:            macOS 13.0.1 (22A400)
Report Version:        12
Anonymous UUID:        1EBFB655-04B0-44EA-A3AF-08981E821F8D

Time Awake Since Boot: 100 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000408
Exception Codes:       0x0000000000000001, 0x0000000000000408

VM Region Info: 0x408 is not in any region.  Bytes before following region: 140737487461368
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      shared memory            7ffffff26000-7ffffff27000 [    4K] r-x/r-x SM=SHM  

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   QtWidgets                     	       0x10c4e5b99 QWidgetPrivate::hideChildren(bool) + 185
1   QtWidgets                     	       0x10c4e72bd QWidget::event(QEvent*) + 4285
2   QtWidgets                     	       0x10c5f7bf5 QMainWindow::event(QEvent*) + 277
3   QtWidgets                     	       0x10c4aa67d QApplicationPrivate::notify_helper(QObject*, QEvent*) + 269
4   QtWidgets                     	       0x10c4aba82 QApplication::notify(QObject*, QEvent*) + 594
5   QtCore                        	       0x10d765324 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
6   QtWidgets                     	       0x10c5076a9 0x10c49a000 + 448169
7   QtWidgets                     	       0x10c504924 0x10c49a000 + 436516
8   QtWidgets                     	       0x10c4aa67d QApplicationPrivate::notify_helper(QObject*, QEvent*) + 269
9   QtWidgets                     	       0x10c4aba82 QApplication::notify(QObject*, QEvent*) + 594
10  QtCore                        	       0x10d765324 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
11  QtGui                         	       0x10ca6e361 QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) + 1409
12  QtGui                         	       0x10ca510e3 bool QWindowSystemInterfacePrivate::handleWindowSystemEvent<QWindowSystemInterface::SynchronousDelivery>(QWindowSystemInterfacePrivate::WindowSystemEvent*) + 115
13  libqcocoa.dylib               	       0x10c14979c 0x10c12d000 + 116636
14  QtCore                        	       0x10d76f6bf QMetaMethod::invoke(QObject*, Qt::ConnectionType, QGenericReturnArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument) const + 1295
15  libqcocoa.dylib               	       0x10c14f1d0 0x10c12d000 + 139728
16  CoreFoundation                	    0x7ff81d4886d5 __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 120
17  CoreFoundation                	    0x7ff81d521ccc ___CFXRegistrationPost_block_invoke + 86
18  CoreFoundation                	    0x7ff81d521c23 _CFXRegistrationPost + 534
19  CoreFoundation                	    0x7ff81d45be61 _CFXNotificationPost + 809
20  Foundation                    	    0x7ff81e1cef2c -[NSNotificationCenter postNotificationName:object:userInfo:] + 82
21  AppKit                        	    0x7ff820e442a9 -[NSWindow(NSWindow_Theme) _finishMinimizeToDock] + 338
22  AppKit                        	    0x7ff8206564ea _NSCGSDockMessageReceive + 507
23  HIToolbox                     	    0x7ff8272df26a DockCallback(unsigned int, unsigned int, void*, void*) + 1961
24  HIServices                    	    0x7ff822cc8d38 _DCXTransactionComplete + 34
25  HIServices                    	    0x7ff822cc8cf5 _XTransactionComplete + 39
26  HIServices                    	    0x7ff822cba5a9 mshMIGPerform + 182
27  CoreFoundation                	    0x7ff81d493418 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE1_PERFORM_FUNCTION__ + 41
28  CoreFoundation                	    0x7ff81d493357 __CFRunLoopDoSource1 + 536
29  CoreFoundation                	    0x7ff81d491fd6 __CFRunLoopRun + 2749
30  CoreFoundation                	    0x7ff81d490e9f CFRunLoopRunSpecific + 560
[... 363 lines cut ...]
      },
      "deploymentId" : 240000014
    }
  ],
  "experiments" : [

  ]
}
}

Model: iMacPro1,1, BootROM 1916.40.2.0.0, 4 processors, Unknown, 2.5 GHz, 3.91 GB, SMC 
Graphics: Display, 7 MB
Display: Unknown Display, 1920 x 1080 (1080p FHD - Full High Definition), Main, MirrorOff, Online
Memory Module: DIMM 0, 3.91 GB, RAM, 0 MHz, QEMU, Unknown
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: QEMU HARDDISK, 402.7 MB
Serial ATA Device: QEMU HARDDISK, 2.54 GB
Serial ATA Device: QEMU HARDDISK, 274.88 GB
USB Device: USB20Bus
USB Device: QEMU USB Tablet
USB Device: QEMU USB Keyboard
Thunderbolt Bus: 
```
</details>

elecpower, 2022-12-13:
> @elecpower SDL2 is the easiest part ;-) I've already tried it out, the patch is minimal. The real question is: do we somehow have a choice? I'm not quite sure we do. Do you know which parts would be impacted the most? Which components are 100% different?

I'll have to set up a Ubuntu Docker toolchain and see what breaks on a Qt compile for starters. 
5.15.2 wasn't too bad with depreciation warnings however that's not the whole story if going to 6.4.

philmoz, 2023-12-13:
Not seeing this anymore with 2.9 or 2.10. Tested on MacOS Ventura (both Intel and M2).
Can probably be closed.

pfeerick, 2023-12-13:
Sounds good to me... can be revivied if it surfaces it's ugly head again.

closed 2023-12-13 by pfeerick