EdgeTX/edgetx #2829: Global variables

trexcarey, 2022-12-08:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Hi, I am trying to set up a complicated mix and overrides for a throttle hold and idle point on a nitro-scale helicopter. I currently adjust the GV with potentiometer S1. I would like to be able to use this global variable within the special functions section. However, with override selected, you can't currently set a value as a GV. Further to this, ideally, upon release of the throttle hold, the throttle should return to the GV value; however, it is unclear how you can set a variable weight to the channel so that this is possible.

### Describe the solution you'd like

It would be good to see GV available in all sections of the mixing software, with the ability to perhaps set variable curve points based on a GV ie. a 2-point curve with the variable GV setting the lower point. The curve could then be made available to set the travel on a channel via the mixes. Furthermore, it could be advantageous for a new GV value to be set with the ability to use of a simple mathematical formula ie. set GV2=GV1-2 etc.

### Describe alternatives you've considered

1. Overrides have been considered, but you cannot set an override as a GV.
2. Mixes have been used with some success, but upon release of the throttle hold, the channel returns to the set weight position. Therefore again, the ability to use GV to set a variable curve would help, as this could be applied through the mixes.

eshifri, 2022-12-08:
Can't you use MAX as a value and a GV as a weight in the overwrite?

trexcarey, 2022-12-15:
No, I can't use Max as I need the idle point to remain the same in both the throttle hold and the throttle idle position governed by the GV. If you use Max, the system returns to -100 when you come out of the throttle hold position into the throttle idle position. GV is currently part of a curve function to only allow small variants over the full range of the potentiometer. Obviously, I could use a curve on the throttle idle position, but that would mean I have two variables to change, one of which I need to go into the radio configuration to adjust. If the GV function could work in SF & curves, this would then mean there is one point of change, and when you move from one function to another, the position would always be the same and adjustable via the same pot. It may be possible with some other programming, but I have yet to find anyone to help combine these functions.

3DRacer, 2022-12-16:
IDoes that apply to my discussion from last year?
https://github.com/EdgeTX/edgetx/discussions/1191

trexcarey, 2023-02-20:
Dev's, is this enhancement going to be looked at? It would seem that being able to access Global variables within all areas of EdgeTx should be reasonably easy to implement within the code. I feel that the team are missing a trick by not having access to these variables in all areas and that this would help with complicated mixes especially with general helicopter and large-scale model programming. I must thank the Devs for their continued implementation and efforts to improve the EdgeTx environment, and hope the above can be added to the list of enhancements in the near future.