EdgeTX/edgetx #2833: Radio settings blank owner registration id corrupt when loaded from radio.yml

elecpower, 2022-12-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Loading a radio.yml file with a blank owner registration id displays junk in the ui.
Companion
[image]

Companion generated
[image]

TX16S
[image]

X9D+ 2019
[image]

Tested with TX16S and X9D+ 2019 simulators and issue on both.

### Expected Behavior

Display saved value

### Steps To Reproduce

From Companion, Simulate Radios.

After editing the field in the radio ui by deleting all the characters and saving, a manual inspection of the radio.yml file shows the label and value the same as generated by Companion.

The radio ui edited blanked value is displayed as blanks on re-entry to the radio settings.

The field value is not padded as entering a single character only stores "x" in the radio.yml file

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2022-12-10:
I believe this to be intentional, not a bug - the UUID from the CPU is used if no ownername is specified - most likely as this is used by the bind/registration process in ACCESS, so _cannot_ ever be left blank. 

https://github.com/EdgeTX/edgetx/blob/2a882c8499595e48953a93951c384050702e7d5f/radio/src/opentx.cpp#L286-L294

elecpower, 2022-12-10:
That would explain why what I thought was uninitialised memory was consistent across sims. I am happy to close based on your evidence.

closed 2022-12-10 by elecpower