EdgeTX/edgetx #2838: Special characters umlauts DE - German missing

koaxheli, 2022-12-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Umlauts for Ä, ä, Ö, ö, Ü, ü are missing again with EdgeTX v 2.8.0
[image]

### Expected Behavior

As with EdgeTX 2.7.0
[image]

### Steps To Reproduce

Special characters are missing in LUA scripts.

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

raphaelcoeffic, 2022-12-10:
@koaxheli you need to compile the german firmware to have these characters (`-DTRANSLATIONS=DE`).

koaxheli, 2022-12-10:
I have made on
https://gitpod.io/#https://github.com/EdgeTX/edgetx/tree/2.8

cmake -Wno-dev -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DHELI=YES -DTRANSLATIONS=DE -DBLUETOOTH=YES -DINTERNAL_GPS=YES -DCMAKE_BUILD_TYPE=Release ../

koaxheli, 2022-12-10:
But the umlauts will not be created as it was done with the first
version of EdgeTX 2.5.

gagarinlg, 2022-12-10:
> I have made on https://gitpod.io/#https://github.com/EdgeTX/edgetx/tree/2.8
> 
> cmake -Wno-dev -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DHELI=YES -DTRANSLATIONS=DE -DBLUETOOTH=YES -DINTERNAL_GPS=YES -DCMAKE_BUILD_TYPE=Release ../

You have to use the correct unicode code points from 2.8 on

gagarinlg, 2022-12-10:
The complete UI on the color LCD radios is now unicode, instead of the pita the old chracter encoding was. With future radios we will then be able to have multi language firmwares

gagarinlg, 2022-12-10:
Did you try using utf-8?

koaxheli, 2022-12-11:
Where and how to use the utf-8 special character set
is called.

gagarinlg, 2022-12-11:
You can just use a modern editor, like notepad++, set it to utf-8 and then enter the umlauts.
This should just work

koaxheli, 2022-12-15:
Thanks, for the tip with UTF-8 encoding has worked with the special characters.

closed 2022-12-15 by koaxheli