EdgeTX/edgetx #2844: Radio Crash to save mode,  when opening internal or external module settings

ParkerEde, 2022-12-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If you have five models with maximum input of the model name (15 characters) and they all have the same ModelID, this leads to a crash when you click on the button "Internal Module" or "External Module" in the model configuration.
This happens in the simulator as well as in the radio firmware. It seems to be related to the determination of the uniqueness of the ModelID.

### Expected Behavior

Radio should not crash.

### Steps To Reproduce

1. create 5 models with the max. model name (15 characters)
2. give all models the same receiver ID with the same RF module protokoll
3. open one of these models in the configuration and try to open the corresponding modulesettings
4. now comes the crash

### Version

2.8.0

### Transmitter

FrSky X12

ParkerEde, 2022-12-10:
here 5 Models with externel MULTI and ModelID = 0 to reproduce it easy
[MODELS.zip](https://github.com/EdgeTX/edgetx/files/10201035/MODELS.zip)

gagarinlg, 2022-12-10:
Thank you for the detailed description and the files.

PR #2845 (2022-12-14): fix(color): EM triggered when checking if modelID unique in certain instances

Fixes #2844

Summary of changes:

In ModelsList::isModelIdUnique function:
 - cast comparison value to int to avoid signed / unsigned comparison error if (curr - warn_buf) > 48.

Note: only tested in the simulator; but should fix issue in affected radios.