EdgeTX/edgetx #2849: Tx reboot & Loss of control

kitsen13, 2022-12-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

TX powered on, commands to the servos OK.
Remove the SD card.
TX message "No SD CARD".
Commands to the servos OK.

Push SD card again into the TX, "No SD CARD disappear OK, controls of servos OK. TX seems  working "normal".
Activate logs recording -> CRASH -> TX reboot -> Emergency mode ON **and loss of control** , no command to the servo are possible.

### Expected Behavior

1) No crash after log record. No reboot of the TX. 
2) Allways maintain control of servo depending of the state of the SD card.
3) If possible : Bip in emergency mode to alert a defective SD card

### Steps To Reproduce

1) TX powered on
2) Remove the SD card.
3) TX is switched to Emergency mode 
5) Push the SD card into the TX
4) Activate logs

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

[https://www.youtube.com/shorts/D2zD5t_Iwd8](url)

kitsen13, 2023-03-03:
Hi, this lead to loss of control if you are using log record like me. A writing problem on the sd card and you can’t recover the model. One rc plane crashed because of a problem on the SD.
 Please can you look at this?

mha1, 2023-05-24:
@kitsen13 Did you use one of the internal MPM protocols to control the model?

kitsen13, 2023-05-24:
> @kitsen13 Did you use one of the internal MPM protocols to control the model?

yes

mha1, 2023-05-24:
I came across the same problem testing recovery from hard faults when using MPM (issue #3599). This was solved in 2.9 with PR #3603 and will also be included in the upcoming 2.8.5. Would be great if you'd repeat your steps to reproduce your problem with this 2.9 firmware: https://github.com/EdgeTX/edgetx/suites/12945731034/artifacts/699714034. This firmware should allow you to regain control after the EM reboot (your point 2. above).

kitsen13, 2023-06-25:
Hi,
Thanks for the reply, i will test this on 2.9 when it will be out. I have planned to migrate 2.8.0 to 2.9.0.

closed 2023-09-05 by raphaelcoeffic