EdgeTX/edgetx #2852: EdgeTX should ignore files with hidden attributes stored on SD card - Feature request in ETX 2.7/2.8

Martivip, 2022-12-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Operation systems like MacOS may add auxiliary information associated with the files "._files, known as "extended attributes" this is how the system works, those files has hidden attributes. The problem is that from Lua perspective menu and from SD card content menu those "extended attributes" files are visible and that causes messy look and is confusing for a lot of users. Problem don't exists when SD card content is created on Windows because Windows don't create those "extended attributes" files.

### Describe the solution you'd like

Edge TX 2.7/2.8 should respect the "hidden" attribute of the files stored on SD card and should not present those files.
It could be also resolved as global setting and the users can on/off visibility of the hidden files.

HEWR1975, 2022-12-17:
Agree, absolutely annoying

gagarinlg, 2022-12-17:
> Agree, absolutely annoying

Yes, absolutely annoying that MacOS does this.

pfeerick, 2022-12-18:
2.7 is unlikely to be changed since 2.8 has since been released. Is this in relation to a colour screen radio, as it appears that B&W radios already hide hidden and dot filename files (i.e. Linux hidden files, so presumably should hide mac files following the filename spec). If so, this means color lcd needs to be taught the same tricks.

gagarinlg, 2022-12-18:
It is a one linet in the file browser. I just don't have any means to test it with an Apple device

pfeerick, 2022-12-18:
Yeah, it actually looks like only the hidden attribute test is missing...

colorlcd only has
https://github.com/EdgeTX/edgetx/blob/73e240f3654f730d16a82735459de521b7ae1d02/radio/src/gui/colorlcd/file_browser.cpp#L148-L149

vs B&W
https://github.com/EdgeTX/edgetx/blob/73e240f3654f730d16a82735459de521b7ae1d02/radio/src/gui/common/stdlcd/radio_sdmanager.cpp#L526-L527

... but this does suggest that both B&W and colorlcd shouldn't show the extra macOS noise such as `.DS_Store`

Martivip, 2022-12-18:
> Yeah, it actually looks like only the hidden attribute test is missing...
> 
> colorlcd only has
> 
> https://github.com/EdgeTX/edgetx/blob/73e240f3654f730d16a82735459de521b7ae1d02/radio/src/gui/colorlcd/file_browser.cpp#L148-L149
> 
> vs B&W
> 
> https://github.com/EdgeTX/edgetx/blob/73e240f3654f730d16a82735459de521b7ae1d02/radio/src/gui/common/stdlcd/radio_sdmanager.cpp#L526-L527
> 
> ... but this does suggest that both B&W and colorlcd shouldn't show the extra macOS noise such as `.DS_Store`
> 
> So perhaps more detail is needed - is this in relation to the SD browser, or something else - and if so, what exactly?

@pfeerick `.DS_Store` are not visible because see that last MacOS Ventura don't creates this files od SD Card but creates duplicates of folders and the files with the naming convention started as `._` files 

[image]
[image]

HEWR1975, 2022-12-18:
The file extrnsion are probably still the same .jpg or .png at the end. Or .lua for that matterGroetjes,Jeroen robroek***********************Vaalserhaagweg 86291CN Vaals+49 157 86766419***********************Sent by Hal

https://github.com/EdgeTX/edgetx/blob/73e240f3654f730d16a82735459de521b7ae1d02/radio/src/gui/colorlcd/file_browser.cpp#L148-L149
https://github.com/EdgeTX/edgetx/blob/73e240f3654f730d16a82735459de521b7ae1d02/radio/src/gui/common/stdlcd/radio_sdmanager.cpp#L526-L527

PR #2925 (2022-12-23): fix(ui) - Ignore hidden file when iterating SD card contents.
Fixes #2852 

Summary of changes:

Add hidden file filter to various places where the SD card files and folders are being iterated over.
closed 2022-12-23 by pfeerick