EdgeTX/edgetx #2853: Overall performance of the 2.8

Martivip, 2022-12-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

General performance of the 2.8 should be improved because is much more slower than on 2.7.1.
In example:
ELRS Lua script loading twice much longer, changes in the LUA also requiring twice much time than on 2.7.1.
Scrolling through the SYS and Model menus is laggy and less responsive.
It would be nice to improve above efficiency 

### Describe the solution you'd like

It would be nice to improve efficiency of the 2.8 that we had on 2.7.1

MrThing, 2022-12-13:
Slow is not actionable. Find the issue. Report that.

Martivip, 2022-12-15:
@MrThing I don't report the issue but enhancement request.

gagarinlg, 2023-05-05:
The menu speed has been improved. Please check the current nightlies.

Martivip, 2023-05-06:
@gagarinlg Thanks definitely I will try

Martivip, 2023-05-07:
@gagarinlg ok I have test it, is better now but still LUA not loads fast as was on 2.7. the same for overall menus performance
My assumption is that lost of the performance is mainly related to LVGL implementation on 2.8, we need faster processors in our radios right  :)

gagarinlg, 2023-05-07:
Yes, for LUA have will expand the API to use LVGL natively, which should speed things up a lot. There is a bunch of stuff we can and will do to further improve the speed, but there are limitations to that.
The big plus with LVGL is that we can do UI stuff much easier now, especially with different screen resolutions and formats without pixel exact tuning everything.

Martivip, 2023-05-07:
@gagarinlg Ok i closing this issue. There is so much goodness in 2.8 and Honestly speaking I'm already use to it :)

closed 2023-05-07 by Martivip