EdgeTX/edgetx #2858: Radio T16S Future Features

AngelDesire23, 2022-12-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

 i like to do a long range flight with my co pilot. During of so many OSD can be shown on Radio T16S , i request a mirroring Feature so T16S Radiomaster can do a Mirroring Screen to Monitor . It will be very helpfull for a co pilot do a monitoring during a long range flight

### Describe the solution you'd like

During of so many OSD can be shown on Radio T16S , i request a mirroring Feature so T16S Radiomaster can do a Mirroring Screen to Monitor . It will be very helpfull for a co pilot do a monitoring during a long range flight

### Describe alternatives you've considered

During of so many OSD can be shown on Radio T16S , i request a mirroring Feature so T16S Radiomaster can do a Mirroring Screen to Monitor . It will be very helpfull for a co pilot do a monitoring during a long range flight

pfeerick, 2022-12-14:
Do you mean you want to be able to have OSD elements that you see on your goggles displayed on the screen of your T16S (i.e. telemetry values)? Or do you mean for your goggles to also be displayed on the T16S screen as well? 

There is no way the T16 will be able to show the image from your goggles AND still be a controller - even if you did figure out some way to connect the goggles to the transmitter in a meaningful way.

gagarinlg, 2023-05-05:
The hardware does not support connecting additional screens

closed 2023-05-05 by gagarinlg