EdgeTX/edgetx #2859: Min/Max indicator in Outputs tab misbehaving

mha1, 2022-12-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The turning point for highlighting Min or Max depends on the stick position at the time of entering the OUTPUT channel (by hitting Edit) 

### Expected Behavior

Min Max indicators should toggle at stick neutral

### Steps To Reproduce

1. Create a simple model using the sticks as inputs for simple mixers (all standard +/- 100)
2. Go to the OUTPUTS tab
3. center all sticks
4. select an OUTPUT channel line that is connected with a self centering stick
5. move stick and check Min/Max indicators changing at around 0%
6. exit the OUTPUT channel line 
7. Hold stick half between neutral and max (e.g. +50%)
8. select same OUTPUT channel line while holding the stick off neutral
9. move stick and observe Min Max changing at around +50%

see video using throttle:

https://user-images.githubusercontent.com/5615068/207436435-c3dc327f-5e0c-435a-a718-ff78114e3c6d.mp4

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

mha1, 2022-12-15:
Thank you very much!

pfeerick, 2022-12-18:
Thank you for the detail report! :)

closed 2022-12-22 by pfeerick