EdgeTX/edgetx #2861: EL18 cannot achieve 5.25M BAUD for CRSF

richardclli, 2022-12-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Can only achieve 1.87M BAUD, error occurs when trying to use 3.75M BAUD or 5.25M BAUD.

### Expected Behavior

Hardware is similar to TX16S, both uses USART2, it should works the same way as TX16S

### Steps To Reproduce

1. External Module
2. Select CRSF
3. Select BAUD to 3.75M BAUD or 5.35M BAUD
4. ELRS will report connection problem.

### Version

2.8.0

### Transmitter

Flysky EL18

richardclli, 2022-12-14:
I think it is because it uses its own telemetry driver, should unify to horus driver to support same features.  Will work out a PR soon.

richardclli, 2023-01-03:
@pfeerick I think this need to be fixed in 2.8.1

richardclli, 2023-01-13:
I will work out another PR forv2.8 branch

richardclli, 2023-01-13:
Added a PR for branch 2.8

richardclli, 2023-02-24:
@pfeerick I think this can be closed, as the new serial driver solves this problem in main and 2.8 already has the fix applied.

pfeerick, 2023-02-24:
Ah yes, thanks... yeah, the PR got close since the serial driver changes superceeded it.

PR #3028 (2023-02-11): Unify telemetry driver for horus and nv14 targets for v2.8 branch, fix 2861

Fixes #2861

Summary of changes:
This PR is for 2.8 branch only, there is https://github.com/EdgeTX/edgetx/pull/2862 is for main.
closed 2023-02-24 by pfeerick