EdgeTX/edgetx #2863: Support Flysky FRM303

richardclli, 2022-12-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

FRM303 is currently not working in any radios, seems the drivers are not properly configured.

FRM303 used the same protocol as the IRM301 in the EL18 with the following information:
1. Radio TX used the PPM pin
2. Radio RX used the heartbeat pin
3. UART bit rate is 1.5M BAUD
4. Signaling is non-inverted

I think supporting this should be easy, just tweaks a bit on the init and config.

### Describe the solution you'd like

Please update the init and config for AFHDS3 ext module to support it properly.

richardclli, 2023-01-03:
@pfeerick See if this can put to 2.8.1

pfeerick, 2023-01-03:
I would prefer 2.9, although since this primarily only affects the AFHDS3 driver even if there is a bug/regression here, it will have limited impact?

Were you able to test this on any other radios than the EL18/NV14 to see if it works anywhere else?

rotorman, 2023-01-03:
Also interesting to know would be if FRM302 still functions or not (I do not have it to test it myself).

richardclli, 2023-01-04:
> Also interesting to know would be if FRM302 still functions or not (I do not have FMR302 to test it myself).

No, Raphael replaced the FRM302 code completely because FRM302 uses the old protocol that the state machine is completely different.

However, for best compatibility, I think it may worth to pull back the old code and make FRM302 works again.  And we may need to think how to organize all these Flysky modules:

FRM302: old protocol
IRM303 (EL18): new protocol with module ID = 0x05
FRM303: new protocol with module ID = 0x04

So for completeness, AFHDS3 should provides 3 sub-protocols: FRM302, IRM3003, FRM303

richardclli, 2023-01-04:
> I would prefer 2.9, although since this primarily only affects the AFHDS3 driver even if there is a bug/regression here, it will have limited impact?
> 
> Were you able to test this on any other radios than the EL18/NV14 to see if it works anywhere else?

FRM303 can only works with Flysky radios because of current hardware limitation and FRM303 firmware support.

In details, besides  generic protocols PPM and SBUS, FRM303 supports opensource UART protocol with 1.5M and 115.2k BAUD.  And the signal is "non inverted".  The serial port connection uses the PPM and heartbeat line in the external module header.

And that is why the X10 PCB series radio, with heartbeat line hardware bug and forced inverted signal may not work with current FRM303 firmware.

I already asked Flysky to add another choice in FRM303 protocol that can make use the S.PORT line to communicate in half duplex just like ELRS did.  Before this, I think it will be hard to get FRM303 supported in other non-Flysky radios.

richardclli, 2023-01-04:
And I have already tested the FRM303 works in NV14/EL18 and PL18.

richardclli, 2023-01-07:
TX16S with ACCESS mod (diode fixed) tested using FRM303 115.2k BAUD opensource protocol option, not working properly.

rotorman, 2023-01-07:
Could you check with a logic analyzer or oscilloscope if the issue stems still from the signal on TX16S Heartbeat line, even after the diode mod, or from something else.

For this you could measure the RF_HB track:

[image]

RF_HB track highlighted here (top layer of TX16S mainboard):

[image]

richardclli, 2023-01-13:
TX16S Int module chained a 2.2k resistor to the Heartbeat pin.  Well, I think better wait for FRM303 firmware update before working out TX16S support.

richardclli, 2023-01-13:
@pfeerick I have made another PR for 2.8 branch as well.

xiaoyigg, 2023-01-16:
I want use radiomaster boxer with frm303
that's ok ?

richardclli, 2023-01-17:
> I want use radiomaster boxer with frm303 that's ok ?

Not supported yet.  Need firmware update in FRM303, or hardware mod.

closed 2023-02-27 by pfeerick

Kai2510, 2024-03-04:
> > Also interesting to know would be if FRM302 still functions or not (I do not have FMR302 to test it myself).
> 
> No, Raphael replaced the FRM302 code completely because FRM302 uses the old protocol that the state machine is completely different.
> 
> However, for best compatibility, I think it may worth to pull back the old code and make FRM302 works again. And we may need to think how to organize all these Flysky modules:
> 
> FRM302: old protocol IRM301 (EL18): new protocol with module ID = 0x05 FRM303: new protocol with module ID = 0x04
> 
> So for completeness, AFHDS3 should provides 3 sub-protocols: FRM302, IRM301, FRM303

It would be better if FRM302 can be driven again, and there's an option to select different protocols. OpenTX 2.3.10 supported FRM302, but sadly now it's removed. Could you further explain the reason, the state machine...?

richardclli, 2024-03-04:
> > > Also interesting to know would be if FRM302 still functions or not (I do not have FMR302 to test it myself).
> > 
> > 
> > No, Raphael replaced the FRM302 code completely because FRM302 uses the old protocol that the state machine is completely different.
> > However, for best compatibility, I think it may worth to pull back the old code and make FRM302 works again. And we may need to think how to organize all these Flysky modules:
> > FRM302: old protocol IRM301 (EL18): new protocol with module ID = 0x05 FRM303: new protocol with module ID = 0x04
> > So for completeness, AFHDS3 should provides 3 sub-protocols: FRM302, IRM301, FRM303
> 
> It would be better if FRM302 can be driven again, and there's an option to select different protocols. OpenTX 2.3.14 supported FRM302, but sadly now it's removed. Could you further explain the reason, the state machine...?

It is always possible if anyone want to code this, but since FRM302 is already discontinued and it can never supports ibus2, so it is out of my interest.

PR #3029 (2023-02-02): FRM303 support for EL18, branch 2.8, enhancement 2863

Fixes #2863

Summary of changes:
PR #2875 (2023-02-27): Added support for FRM303, enhancement 2863

Fixes #2863

Summary of changes: