EdgeTX/edgetx #2871: Radiomaster 16s scroll wheel

SQuiDFPV, 2022-12-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Updated to flying Dutchman. And my scroll wheel no longer works, correctly.  It goes up 1 and down one but no further.  Cannot use lua scripts via touch screen so can no longer bind.

At Xmas whoop party, I swapped Someone else's scroll wheel in, and it's the same result. Only happened after edgetx update. 

Will down grade and see if I have the shop wheel back 

### Expected Behavior

Scroll wheel works.

### Steps To Reproduce

Update to latest edgetx

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

SQuiDFPV, 2022-12-15:
Flashed back to 2.7.1 i have my scroll wheel back.

bigspeedfpv, 2022-12-20:
Can't reproduce on mine. TX16s mk1, maybe your firmware is corrupted?

SQuiDFPV, 2022-12-20:
Back on 2.7.1 things are normal.
Not sure,  just passing on what happened.

pfeerick, 2022-12-21:
This is the first report I have heard of this, so I would also be inclined to think it's corrupt firmware since reverting to 2.7.1 and fixing it is suggestive that it isn't a hardware issue. Be interesting to see if anyone else encounters this, or it happens again.

rotorman, 2022-12-21:
Tested the official v2.8.0-release on TX16S mkII. I did not encounter any scroll wheel issue.
@SQuiDFPV Possibly you have:
a) wrong fw-target loaded, try reflashing in DFU mode with EdgeTX Buddy: https://buddy.edgetx.org/#/flash?version=v2.8.0&target=tx16s
b) smth. wrong with your SD card content. Grab another card and use again Buddy to make you a fresh EdgeTX SD card to try with: https://buddy.edgetx.org/#/sdcard
c) if a anb b still do not lead to success, smth. wrong with the hardware. Have a look at keys diagnose page and the R.E. (rotary encoder) value. Possibly make a video when you move the roller and post the link here to have a better look of what is going on.

Jeremyforfilm198, 2023-01-07:
I am currently having the exact same issue with the scroller on my TX16S w/EdgeTX. I updated the FW to EdgeTX2.8.1 and still no success. I am going to try what you did and downgrade to 2.7

I am posting this because I wanted to share that this is a shared problem.

rotorman, 2023-01-07:
@Jeremyforfilm198 The latest release presently is still v2.8.0 https://github.com/EdgeTX/edgetx/releases so dunno, where did you get v2.8.1?
Are you sure you flashed TX16S fw and NOT T16 fw? What does the SYSTEM -> VERSION page list for you (can you post a photo/screenshot)?

Jeremyforfilm198, 2023-01-07:
That was a typo. It’s actually just 2.8.0

I just tried to downgrade to 2.7 but it’s still not working (scroll wheel)

rotorman, 2023-01-07:
You did not provide a photo/screenshot of your SYSTEM -> VERSION page?
In case you flash the correct TX16S firmware and the scroll wheel is not working, there is likely a hw defect. But, before we try to fix that, please provide the info from the VERSION page.

Jeremyforfilm198, 2023-01-07:
I think I found the problem. There’s a tear in the ribbon cable.

 I ordered a replacement and will post if it worked after I install it.

Appreciate you!

closed 2023-05-05 by gagarinlg

gazhayes, 2024-03-24:
I have the same issue after updating from openTX 2.3.6 (Jumper T16). Created a separate issue.

The latest version of OpenTX has the same issue. Unfortunately I can't revert to 2.3.6 because the files are no longer available.

Any suggestions?

pfeerick, 2024-03-24:
You could try flashing 2.7.1, and see if that has the same effect the OP mentioned. Otherwise, I would be thinking of doing the same tests Risto mentioned above (via the Radio Settings -> Hardware -> Analogs page) and posting the results in your TX16S issue also. i.e. checking what the firmware says the rotary encoder is doing.

3djc, 2024-03-24:
He said Jumper T16 I think, where there is a number of things (ribbon) that could go wrong

gazhayes, 2024-03-24:
Ok I've flashed 2.7.1 and now instead of going back and forth between two menu items I can make it move around more by flicking the wheel faster (this was not possible with 2.9.x).

It was 100% working with 2.3.6 so it's probably not strictly a hardware issue.

On the key diagnostics page, the rotary encoder value alternates between `0` and `2` with each click of the wheel. What *should* it be doing? 

Based on my arduino experience it really feels like a debounce timing issue.

pfeerick, 2024-03-24:
This reminds me of a discussion about debouncing in software... and that it actually isn't needed if the code is written properly (or done properly in hardware for that matter)... i.e. like https://github.com/buxtronix/arduino/tree/master/libraries/Rotary ... and normally you can tell if it is debouncing if slow turns work fine, and fast ones skip or 'jitter' in place... like the one on an ALDI microwave that I have :confounded: 

Anyway, it *should* be counting up or down consecutively as you keep scrolling up or down. I don't know if the rotary encoder handling changed between opentx 2.3.6 and 2.3.15 , but I now it has changed a little bit in ETX over the years... and it is possible it exposed a underlying hardware issue related to the seating of the ribbon cable between the mainboard and rotary encoder sub-board. Please do note that while it does have hardware debouncing/filtering (220 Ohm resistor and 100 nF cap) 

One thing to be aware of, is that the T16 (and also first gen TX16S) have the rotary encoder filtering mis-wired, so there is actually some special code to work around that... and it's a miracle it works at all. This is the relevant section of the schematic from the TX16S (thanks to Risto's [reverse engineering efforts](https://www.rcgroups.com/forums/showpost.php?p=47115381&postcount=3)):
[image]

oh., how much trouble putting a capacitor on the wrong side of a resistor can cause (`ROTARY_ENCODER_SUPPORT_BUGGY_WIRING`)! :open_mouth: 

https://github.com/EdgeTX/edgetx/blob/f7f886095a03c8e3a735be67a31ab78b0d1b954a/radio/src/targets/common/arm/stm32/rotary_encoder_driver.cpp#L60-L139

pfeerick, 2024-03-27:
Thanks, corrected. One of reasons I shouldn't read schematics just before bed... read the two switches as the two switches of the encoder and didn't look at the net names. 

This is somewhat misleading/confusing then, isn't it? The filtering is in relation to the buttons (which you discuss later), not the encoder? 

> The roller encoder is using [Alps EC10E](https://tech.alpsalpine.com/prod/e/pdf/encoder/incremental/ec10e/ec10e.pdf) and it basically functions as two combined switches in phase shift (quadrature encoding). **The filtering is similarly to trims and switches with a 220 Ohm resistor and 100 nF cap, but here the parts are populated directly onto the roller board assembly**. Unfortunately RadioMaster did not wire the encoder up correctly. It is apparently similarly wrongly wired also in the Jumper T16, whereas in FrSky Horus X12 it is wired according to ALPS datasheet.

rotorman, 2024-03-27:
I see. Indeed the highlighted sentence is talking about encoder button and not quadrature encoder lines, which only have current surge limiting 4.7 kOhm resistors, and no caps are involved. RC filtering with 220 Ohm resistor and 100 nF cap is only on the buttons.

PercyJax, 2025-01-29:
Sorry to revive an old thread, but there may be a regression: I've just purchased and upgraded a TX16S MKii (whose rotary encoder worked on whichever version of the firmware it came with) and upgraded to both 2.10.5 and 2.10.6 and it exhibits the same symptoms (can only scroll back and forth one item)

PercyJax, 2025-02-12:
In case anyone else also arrives here from frantic searching, and experiences the same issue (where the hardware debugger just gives you values like 0 <-> ±1 or 0 <-> ±2 no matter how you scroll, the issue is likely not the scroll wheel assembly or radiomaster's circuit design mistake, but rather something wrong relating to the ribbon cable between the main board and breakout. I didn't have a replacement one but I just swapped the ends and the difference in curvature solved my issue for now. 

Lots of red herrings here, since the code changed around the rotary encoder between 2.7.1 and 2.10.6, I wasted a lot of time digging on that front; the older code seemed to have had a tiny bit more success interpreting the crappy signal it's getting from the encoder but just save yourself the headache and check the ribbon cable first. 

Thanks @rotorman for the tip.