EdgeTX/edgetx #2881: No CH6-output: Missing CYC-entries at CV32 pointer

ReneNwnhs, 2022-12-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In Companion i setup a heli with a 120-y swash. After entering all the information i had, i decided to turn it into a template without ccpm (two blade swash) so i unchecked the box in the heli-tab, as the mixing was no longer needed. Later on i tried to finish the setup in the TX, but could not get the output on CH6 (collective pitch) moving. It (the output) appeared to have a link to CV32, the last (empty) line of the curve-table, and the last before the CYC's that should have been gone). Those CYC's still appeared to be selected in several dropdown-boxes i noticed.

Regards,

René

### Expected Behavior

Would expect that CYC's (1,2 and 3) would be gone and no longer selectable and reference in output-table deleted.

### Steps To Reproduce

Create a heli-model with a 120-Y swash
Finish the setup
In the heli-tab remove the swash-selection
Try to get the collective pitch to move

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

ReneNwnhs, 2023-01-04:
I did some reading on github today an ran into this: https://github.com/EdgeTX/edgetx/issues/1092. Does the discussion of the Heli-issues lead to the conclusion that support for Heritage-Heli's is dropped? I aggree with some of the perticipants it is not so hard to solve, and would generate a lot of credit. All you have to do is look what a the collective servo does on a heli with two servo's (elevator and aileron) on the swish. The the collective servo does is sliding the swash up or down along the mainrotor-axis and thus moving the offsetpoint for the elevator and aileron movements. That limits the fysical movement of these, but that can be calculated. The total movement is defined (about 25 degrees; 12,5 up and down) the pilots choice is howmuch of this to use for collective (rocketing or ballooning up/down) and how much for ele/ail (bowling ball, or lazy roll).

ReneNwnhs, 2023-05-05:
> No, nothing has been dropped. 2.8 just had not HELI enabled by default
> and you need a HELI enabled build. With 2.9 you will get an option to
> enable/disable the HELI tab in the model configuration without the
> need for a different firmwarre

Hi Malte,

Thanks for the answer. As a participant (René N) on John Hopke's Discord 
channel (RCVR) I read your and Peter's answers on this.

As to the CV32 curve: on most of my models this curve appears in the 
outputs table on channels 1, 2 and 4 (ail, ele and rud). Is this created 
by a wizzard at some point? And is it useless in a completed set-up? Or 
has is a function although its linear and I could not find any further 
connections or links.

Regards,

René

Ps changed domain.

gagarinlg, 2023-05-05:
It is probably an old problem. I suspect nobody expected that someone would configure a complete model with heli-mixers and then switch them off.
Can you upload on of those model files, so that I can take a look?