EdgeTX/edgetx #2897: LUA extension to model object to retirve the fileName where the model information is stored

frankiearzu, 2022-12-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have been writing LUA scripts to help on the configuration of DSM Forward programming tools.
(See Muti-Module DSM Forward Programming v051).
During the development, i need to persist additional model specific configuration related to Spektrum model setting (Aircraft type, Wing Type, TailType, channel mappings, etc) 

My solution was to create a file in my tool data directory, but i need a unique name. So i was looking to use the same filename as the model is store in /MODELS.  Looking at the model.getInfo() return object/table. I can get the display name (non unique), as well as the bitmap, but could not find a way to get the unique file name.

model.getInfo()
```
table model information:
name (string) model name
bitmap (string) bitmap name (not present on X7)
```

### Describe the solution you'd like

Option 1:
Add the model fileName information to the model.getInfo() table.  In that way, it will be backward compatible API, and will enable external tools to have a way to create files in other directories associated to that model with a name that will not change, even when changing the display name.

Option 2:
Allow a way to store in the model file additional custom information for external tools.

pfeerick, 2022-12-19:
Appologies, it was an oversight in the documentation - filename was actually added to the table returned from that function in 2.6.0, so is available from then onwards. [Documentation has been updated](https://luadoc.edgetx.org/part_iii_-_opentx_lua_api_reference/model-functions-less-than-greater-than-luadoc-begin-model/getinfo) to match the code.

closed 2022-12-19 by pfeerick

frankiearzu, 2022-12-19:
Great! Thank you Peter.  Once I'm done with helping on the DSM forward programming tool (in Multi-Module) will look on what to help here.  In there, i have to write a model configurator similar to how Spektrum radios since the RX needs that info during the setup.. yes, the RX needs to knows the servo assignments and mixes (wing type, tail type) to properly drive the Gyro/AutoLevel/Panic. (nor programable via Forward Programming menus, but the TX pass that info behid the scenes to the RX)
Frankie