EdgeTX/edgetx #2898: Mame the Model creation wizards more usefull by allowing to edit model who was created with it before.

frankiearzu, 2022-12-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The model wizard is a one time setup scripts.  It could be nice to be able to go back to it to edit the current model  (change servo settings, wing type, tail type, etc).   If we persist that information somewhere, the wizard can load the previos known information to modify it.
This is also related to my other request #2897

Using Futaba and Spektrum radios before, is much easier to create models and edit them that way.. that will make it easier for newbies to setup and edit models.

Additionally, storing that Aircract type, WingType, Tail Type, and channel mapping will allow other tools to use that infomation.
Like the DSM forward programming tool, that needs that info to correctly confugure new Spektrum RXs from scracts; needs servo assigments and wing type mixing, like Vtail, Delta Wing, etc. , currently have to duplicate that Wizard functionality on my script, but could be something that can be used to setup the radio.

Mostly adding this enhancement request to see how much interest and opinions.  I could do the programing, but want to see the opinion of other developpers here.

### Describe the solution you'd like

Evaluate the request, and see if there is something that could benefit EdgeTX to make it easier to use.

lshems, 2022-12-19:
I created a set of 'default' setups with a lua 'wizard' which just does that. Check www.justfly.solutions.

There are also free solutions out there.

I do support the idea, and it should be linked to the existing issues about model interchangability between radios.

I do not like the way the current wizards are set up so I won't help changing them. If model interchangability issues would be solved, I would be open for spending time on a wizard. But it would be more like standard Mia setup, where you pick a model template from a lost and then change it using a wizard.

L'shems.

frankiearzu, 2022-12-19:
L'shems,
Really nice tools.. i am familiar with the complexity of gliders wings and mixes (flew F3B, F3J).
Currently, Im working on DSM Forward Programming tool for Spektrum receivers with Gyro (Stabilization, AutoLevel, Envelope, Panic),
The RX needs to know where are your ailerons (primary, and slave),  elevator/rudder (one or two of each), as well as if you have any special mix for them (Delta Wing, Vtail or Ailevator tail).  In that way the RX can "fly" the plane.

Without a common model Wing/Tail type representation that can be shared across tools/scripts, you will have to setup similar things in multiple different scripts.

I think there are two framework things here:  
1. Metadata to describe the model Wing and Tail, with the ability to setup your own channels/mixers.
2. Tools to do this configuration
2.a) Wizard to setup your servos and mixes on your own (Create as well as Modify)
2.b) Pre-Configured model templates like the JustFly or the RCSoar templates (with servos assigned, and other things).

For example, imagine that JustFly can create the metadata,  The DSM Forward Programming tool can take that metadata and just use it.    Same thing with the common wizards, they can create the metadata, and DSM Forward programming can use it.

Currently, the Gyro script (Spektrum or FrSky) will need to provide its own way to descrive the plane to the gyro!  Double work.
On radios like Spektrum/Futaba, all this is integrated, so you don't need to redo things.

Examples of my Gyro Setup screens: 
![Capture5](https://user-images.githubusercontent.com/326043
![Capture3](https://user-images.githubusercontent.com/32604366/208478
[image]
757-81a862d8-be1b-4a35-ae80-ce1d56acf9bd.JPG)

[image]
66/208478671-15c5f188-52fa-4b69-9940-aebba2c2dc51.JPG)

lshems, 2022-12-19:
You just need naming conventions on the outputs. That's all.

Back to closed system models again, that is on input and output level. In between you can do what you want with the mixers.

Lol

frankiearzu, 2022-12-19:
Yes, discusions would be fine.. I really don't think that we are doing a "close" system again, just helping 
inteoperability between tools.