EdgeTX/edgetx #2899: In companion the sort order and selected labels from the Model Select page are not loaded from the radio models and settings data.

philmoz, 2022-12-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Even though the last used sort order and selected labels are stored in the labels.yml file on the radio they are not used by companion and discarded when the models and settings are loaded from the radio.

When you first load the simulator the model select page has no selected labels and defaults to the alphabetical ascending order.

If you look at the labels.yml file saved to the .etx file it only contains the label names and is missing all the other data stored in this file on the radio.

I'm not sure I know the code well enough yet to fix this so any tips would be appreciated.

### Expected Behavior

The simulator starts with the sort order and selected labels that were last used & saved on the radio.
The labels.yml file saved to the .etx file has all of the data from the radio labels.yml file.

### Steps To Reproduce

1. Set sort order and selected labels on the radio
2. Load models and settings from radio into Companion
3. Start simulator
4. Open model select page - note the radio sort order and selected labels are not used

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2022-12-19:
@dlktdr Was this something you were working on or was it in the "would like but don't have the time to work on" pile?

dlktdr, 2022-12-19:
It was a just get it in for 2.8, not ideal but since it gets re-scanned anyway will work.

Right now it looses the sort order, selected labels and the last opened time, since everything else is re-scanned on any models that were written there wasn't too much point of storing all the other data.

I have a branch I was working on to allow companion to load the sort order and pick in the model browser, same as the radio but got stuck on the proxy model not showing any items after enabling it. https://github.com/dlktdr/edgetx/tree/cpn_add_sort Also corrects the issue if you drag a model in it's own window creates a duplicate. Keeping selected labels should be an easy correction. Haven't had a chance to get back to it yet.

To keep the last opened time a change on the radio is also needed to keep this field. If it detects the file changed this information gets lost on re-scan.

@philmoz I saw your other pull request to correct the not deleting models 👍  looks like you have a good grasp on Qt, want to help plug away at some of these issues together? Are you on discord? There were other issues as well that just got patched to make it into 2.8 that need to be discussed as well.

philmoz, 2022-12-19:
@dlktdr Thanks for the info, I will take a look at your branch and see what I can learn :)
I'm not an expert on Qt although I have a lot of experience with other systems like iOS and Xamarin forms. EdgeTX is pretty complex and I'm still getting my head around it.
Happy to help out if there are things with higher priority - just might need a few pointers along the way.
Yes, I have a discord login but I don't check it all the time.

dlktdr, 2022-12-19:
This branch does need some ❤️. What's your discord handle?

Yeah it's lot to learn, I feel like I've still barely touched the surface. The community here is amazing any insight you need just ask in #dev. Anything your willing to help with is greatly appreciated.

gagarinlg, 2022-12-19:
Please join the dev channel, especially for quick discussions and throwing ideas around, it is easier and quicker than github

elecpower, 2023-04-07:
@dlktdr @philmoz any objections to me taking this one on?

philmoz, 2023-04-07:
@elecpower no objection from me. It was still on my todo list; but had not got around to it (did not seem high priority as the data gets regenerated on the radio).

elecpower, 2023-04-07:
PR #3033 is related so thought of tackling together

dlktdr, 2023-04-07:
> @dlktdr @philmoz any objections to me taking this one on?

No objection, haven't had time in my life lately to work on it anyway.

PR #3580 (2023-05-23): fix(cpn): write sort order and selected labels
Fixes #2899

Summary of changes:
- based on work by @dlktdr 
- extend radio data structures
- read and write sort order and selected labels

Note: This PR does not support changing these within Companion. That will be included in a separate PR along with some other features. The primary objective here is for Companion not to lose settings.
closed 2023-05-23 by pfeerick