EdgeTX/edgetx #2901: Assign extra PCB inputs available already in X9-Lite to SG and LS and RS.

r41065, 2022-12-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

See OpenTx Issue #6555:
I have updated all my transmitters to Edgetx to have the same operating system on all of them.
Waiting in OpenTx that someone works on the additional switches but stale Robot closed that one.
Is there any hope to get this working in EdgeTx ?

### Describe the solution you'd like

The X9-Lite has 5 switches and a pot. On, the board, provisions have been made for easy diy extension with 2 pots (or switches) and one switch (to switch a 2 pos or toggle).

I would suggest to include these already in the available hardware options on the X9-Lite.

### Additional context

https://github.com/opentx/opentx/issues/6555

pfeerick, 2022-12-19:
It should be doable - for the TX12 and TX16 we have support for the extra "unused" ports and it's a matter of configuring what they do. Plus we don't have the storage format issue that existed at the time this issue was mentioned at the time.

r41065, 2022-12-19:
can take some time because I have a job where I am on the road a lot ( only 1 weekend per month at home... )

r41065, 2022-12-19:
[image]
[image]

Pictures are from Mike Blandford,

raphaelcoeffic, 2022-12-19:
Just looked up the datasheet (see [here](https://www.st.com/resource/en/datasheet/stm32f207vg.pdf))

| Pin     | Alternate function |
|-------|-------------|
|`PD14`|`TIM4_CH3`|
|`PC2`|`ADC123_IN12`|
|`PC3`|`ADC123_IN13`|

raphaelcoeffic, 2022-12-19:
@r41065 as it seems, `PC2` & `PC3` can be used as analog inputs, while `PD14` could be used as an additional `2-pos` switch.

gagarinlg, 2023-05-05:
@3djc maybe this can be done easily, after the switch-refactor merge

r41065, 2024-05-16:
one year after last comment, any news?

raphaelcoeffic, 2024-05-19:
@r41065 right, I think we could easily do it now. Anyone who would be able to test?