EdgeTX/edgetx #2917: Flashing Receiver Firmware via External Bay fails (comms error).

Lex-Dysia, 2022-12-21:
Hello, I'm running V2.7.1 on a Radiomaster TX16S. Today for the first time (on EdgeTX) I tried to flash a RadioMaster R168 receiver via the external bay of the radio. I tried 2 different versions of the firmware (.frk files) several times all resulting in failure messages. Protocol was set to PPM.

I then tried the same exercise with the same firmware files and the same receiver on a Taranis running a recent version of OpenTX and everything worked correctly.

Lex-Dysia, 2022-12-22:
> You are correct, this repository is for the EdgeTX Webpage. I will move it to the main EdgeTX Repo.

Doh!! Thanks.

pfeerick, 2022-12-22:
I'll try to update mine again tomorrow to double-check the status of firmware updating the RXs, but there was a detailed investigation into a similar problem about a year ago, which reinforced the pre-existing comments on RCG that that R168s were temperamental to flash - and in some instances had a faulty inverter chip - https://github.com/EdgeTX/edgetx/issues/1069

I don't know if that is relevant here - just something to be aware of.

gagarinlg, 2023-05-05:
I think this is fixed with the serial-refactor merge + the bugfixes made afterwards. @raphaelcoeffic can you confirm?