EdgeTX/edgetx #2921: No jump to main screen after selecting a model

mha1, 2022-12-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Selecting a model that as am associated <modelname>.txt in the MODELS folder does not jump to the main screen after selecting the model. Furthermore, after leaving the select model screen manually the main screen is shown, the <modelname>.txt is not displayed first.

### Expected Behavior

Jump to main screen after selecting a model regardless of <modelname>.txt presence. If <modelname>.txt is present the <modelname>.txt should be displayed first rather than the main screen.

### Steps To Reproduce

1. create simple model
2. add <modelname>.txt to MODELS folder
3. select any other model
4. select newly created model
5. observe system remains in model select screen
6. leave model select screen
7. observe system selected main screen without displaying text first

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

ParkerEde, 2022-12-21:
I can confirm the problem and have reproduced it with the X10S Express and Jumper T18.

ParkerEde, 2022-12-21:
I think it's the same like
https://github.com/EdgeTX/edgetx/issues/2722

eshifri, 2022-12-22:
It is fixed in #2792

pfeerick, 2022-12-22:
Duplicate of #2722

closed 2022-12-22 by mha1