EdgeTX/edgetx #2924: Companion Joystick Calibration Channel Mismatches

jmxp69, 2022-12-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

PC connection via FrSky Sim Dongle.

TX16s 2.7.1 Mixes:
Ch1 - Ail
Ch2 - Ele
Ch3 - Thr
Ch4 - Rud
Ch6 - S2
Ch8 - S1

Configure Joystick Map:
Ch1 - Ail
Ch2 - Ele
Ch3 - Thr
Ch6 - Rud
Ch4 - S2
Ch5 - S1

Using Ch5 in the mixer with movement on output via channel monitor shows no movement in Configure Joystick Dialogue.
Using Ch7 in the mixer with movement on output via channel monitor shows no movement in Configure Joystick Dialogue.

### Expected Behavior

Mapping from radio outputs to companion configure joystick dialogue should be a direct 1:1 map. Ch1=Ch1 .... Chx=Chx

### Steps To Reproduce

Companion 2.8
Frsky sim dongle paired to radio
Add mix lines as described above
Edit Settings on Companion
Simulator Settings
Joystick Calibration
Witness mismatched channels on joystick calibration dialogue compared to mix lines on radio.

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

psychotik2k3, 2023-01-21:
i confirm the bug on my TX16s mk2 ETX2.8.0, at least no movements on channel5 while there is movement on the windows joystick test windows

[image]

pfeerick, 2023-01-24:
Are you in a position where you #2277 a try? I suspect this will fix the channel order, but not necessarily the non-responsive channels (CH7/CH8 in my case with T.A.E.R.S1.S2.LS.RS mixer). 

For me, on Windows 11:
```
T -> CH1
A -> CH2
E -> CH3
R -> CH6
S1 -> missing (but responsive in windows joystick config)
S2 -> CH4
LS -> missing  (but responsive in windows joystick config)
RS -> CH5
```

Note that you should probably also calibrate via the game controllers/joystick control panel first, and only then open up Companion so that it can read the calibrated joystick values - if you try to calibrate it while companion is open it doesn't seem to update.

psychotik2k3, 2023-01-24:
> Are you in a position where you can give #2277 a try? I suspect this will fix the channel order, but not necessarily the non-responsive channels (CH7/CH8 in my case with T.A.E.R.S1.S2.LS.RS mixer).
> 
> For me, on Windows 11:
> 
> ```
> T -> CH1
> A -> CH2
> E -> CH3
> R -> CH6
> S1 -> missing (but responsive in windows joystick config)
> S2 -> CH4
> LS -> missing  (but responsive in windows joystick config)
> RS -> CH5
> ```
> 
> Note that you should probably also calibrate via the game controllers/joystick control panel first, and only then open up Companion so that it can read the calibrated joystick values - if you try to calibrate it while companion is open it doesn't seem to update.

do you have a nightly version that include this PR ? as only firmware is provided as asset of the nightly. thanks

pfeerick, 2023-01-24:
Actually, no every PR has firmware assets (via the Checks tab), but that PR is a bit behind now so I'll have to build it locally, so will upload that when I do and we can both try it then.

psychotik2k3, 2023-01-24:
ok thanks.

sebastien

phileville, 2023-05-14:
I have tested this in  a 2.9.0 nighly with the changes for #2277 and it is working correctly of standard USB connection in classic mode.  All radio channels ( I tested channels 1 thru 11) are mapping 1 to1 in Windows USB joystick test screen and EdgeTX Companion (V2.9.0) simulator joystick configuration window.  It seems to be working for me.  I DO NOT have a FrSky Sim Dongle to test that however.

pfeerick, 2023-05-15:
I did have one single issue with Windows and joystick still post #2277 and #3400 (which further improves things) ... where CH7 and CH8 are swapped in 'Classic' mode. However, this is limited to Windows only - both Mac and Linux are both correct IIRC. And in 'Advanced mode', if you follow the order X,Y,Z,rotX,rotY,rotZ,Dial,Slider (_NOT_ Slider, Dial) it seems to satisfy what seems to be either a bug incompatibility between how the different OSes handle the order of the joystick data. 

With the FrSky Sim Dongle, the RX in the dongle should receive the channels in order in the mixer, so what it does with the channels after that is outside of our control. Any bugs in channel order and handling are on the Frsky/Microsoft side, not EdgeTXs, now that the issues Companion did have seem to be resolved.

closed 2023-05-15 by pfeerick