EdgeTX/edgetx #2927: Crash when opening Hardwaresettings in SYS

ParkerEde, 2022-12-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

The simulator of a TX16S crashes immediately when you try to open the "Hardware" menu in "SYS". Except for the Horus X10S Express, all other ColorLCD radios seem to be affected. Tested with current MAIN.

### Expected Behavior

should not crash

### Steps To Reproduce

1. Start Companion
2. select a RM TX16S profile
3. Create a new document
4. Add a new model (cancel wizard)
5. simulate Radio
6. cancel switch warning
7. open systemmenu (SYS)
8. try to open Hardwaremenu (touch or page button)
9. see the crash

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

2.9.0 selfbuild 8eb55452

philmoz, 2022-12-22:
The simulator doesn't crash for me when built from the current main branch.

Did you build companion and simulator or download from somewhere?
If you built them can you share the settings for your build.

ParkerEde, 2022-12-23:
I build them. I try the actual main now and come back.
This is my configuration:
cmake -G "MSYS Makefiles" -Wno-dev -DCMAKE_PREFIX_PATH=$HOME/5.12.9/mingw73_64 -DSDL_LIBRARY_PATH=/mingw64/bin/ -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DHELI=YES -DLUA=YES -DINTERNAL_GPS=NO -DTRANSLATIONS="DE" -DCMAKE_BUILD_TYPE=Release ../

ParkerEde, 2022-12-23:
I just built this DLL and with it the simulator still crashes on me. Can you reproduce it with this DLL?
[libedgetx-tx16s-simulator.zip](https://github.com/EdgeTX/edgetx/files/10293478/libedgetx-tx16s-simulator.zip)

ParkerEde, 2022-12-23:
When I set -DTRANSLATIONS="DE" to "EN", the simulator does not crash. So it seems to be the translation to German.

ParkerEde, 2022-12-23:
I found the error. All other languages but EN are affected. This line:
#define TR_AUX_SERIAL_MODES            "AUS","Telem weiterl.","Telemetrie In","SBUS Eingang","LUA","CLI","GPS","Debug","SpaceMouse","External module"

was extended with ,"External module" but this is missing in all other language files. I added it to DE and now it works fine.

I think we should do a collection pull request so that each individual language doesn't have to do that separately now. Okay?

DE:
#define TR_AUX_SERIAL_MODES "OFF", "Telem weiterl.", "Telemetrie In", "SBUS Eingang", "LUA", "CLI", "GPS", "Debug", "SpaceMouse", "Externes Modul".
I think we should do a collection pull request so that each individual language doesn't have to do that separately now. Okay?

ParkerEde, 2022-12-23:
And here the PR where only EN was touched, and all other Languages not:
https://github.com/EdgeTX/edgetx/pull/2562/files

pfeerick, 2022-12-23:
That would be right... only 'EN' is automatically tested. I usually catch
any of the non-en bugs when doing language builds :-/

I know that one is needed, and have this suspicion there is something else
as well. Just check the 'en.h' file for recent changes and see if there are
any others, otherwise we just need a followup pr to translate "external
module"

ulfhedlund, 2022-12-23:
I can confirm the same issue for Swedish firmware.

closed 2022-12-25 by pfeerick