EdgeTX/edgetx #2929: Companion win 32 installer and linux appimage missing open ssl libraries

elecpower, 2022-12-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Build system

### Current Behavior

Various reports in issues including #2758 and Discord

### Expected Behavior

Libraries should be bundled into app directory to ensure version compatible with Qt is installed and loaded at runtime

### Steps To Reproduce

Install application on a computer without openssl installed.
Run Companion update process.
Ssl error

### Version

2.8.0

### Transmitter

Other (Please specify below)

elecpower, 2022-12-22:
Pulled apart the 2.8.0 Appimage - not included
Ran the Win 32 bit installer - not included
Ran the Win 64 bit installer - included

elecpower, 2022-12-23:
Windows Companion 32-bit Github action reports "Installer: SSL libs not found, set SSL_LIB_DIR manually." in Build step even though the openssl dependency appears to have been installed. So likely search paths incomplete.

Gerold68, 2022-12-26:
The Linux companion has the same problem. Just Win is fixed.

closed 2023-02-22 by pfeerick

vyalyh-oleg, 2023-05-05:
It still doesn't work for Linux x64 version. Ubuntu/Debian/Mint stable LTS releases.
Mine is:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Linuxmint
Description:	Linux Mint 21.1
Release:	21.1
Codename:	vera

$ uname -a
Linux Sword 5.15.0-71-generic #78-Ubuntu SMP Tue Apr 18 09:00:29 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

$ apt show libssl3 
Package: libssl3
Version: 3.0.2-0ubuntu1.9
```

elecpower, 2023-05-05:
@vyalyh-oleg your OS has an unsupported libssl version. Are you running the release AppImage or a self build?

PR #2935 (2023-02-22): fix(build): packaged openssl files
Fixes #2929 #2758

Summary of changes:
- modify the github workflows including installing Qt tools openssl for each OS
- change the Companion installer processes to bundle the Qt versions of the openssl libs

Dependencies:
For linux builds https://github.com/EdgeTX/build-edgetx/pull/14 needs to be merged

TODO:
- [x] Win x86
- [x] Win x64
- [x] linux
- [ ] MacOS