EdgeTX/edgetx #2933: SerialPort Config in X10SExpress

ParkerEde, 2022-12-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

I'm not sure exactly what would be right here. But between the old X10 and the X10-Express there is the difference that the serial port is naged differently for configuration:
X10:
[image]

X10Express:
[image]

### Expected Behavior

If it is correct that no configuration can be done on the X10Express, then the headline itself should be hidden.
Otherwise, should be made accessible to configuration

### Steps To Reproduce

Open Hardwaresettings on X10 and X10Express and compare

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 / X10S (ACCST), FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

nightly 2.9.0

donkeydog67, 2023-01-05:
Is this a bug? I also have the same issue. Does X10S Express handle it internally? I have no idea

donkeydog67, 2023-01-10:
Does X10S Express handle it internally? Did someone fix the issue?

donkeydog67, 2023-01-17:
Is this bug solved or stil exist?

brainbubblersbest, 2023-01-17:
The serial is used for the express/externalaccess Functionality.
On X10Express Radios straight from Factory.
On that Radio you cant unwire the serial.
But it would make sense to use the serial by connecting Hardware in the HF Bay. 😄

raphaelcoeffic, 2023-02-11:
I believe there should be at least `USB-VCP` as an additional line there, but maybe this is a glitch in the simulation.

ParkerEde, 2023-02-11:
> I believe there should be at least `USB-VCP` as an additional line there, but maybe this is a glitch in the simulation.

yes that's right. On the real radio, there is "USB-VCP"

ParkerEde, 2023-03-16:
USB-VCP is now in Simu. So we can close here

closed 2023-03-16 by ParkerEde