EdgeTX/edgetx #2934: flashing a firmware into an Archer Rx is extremely slow (via ext. port)

ParkerEde, 2022-12-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I flash the firmware of an Archer receiver with EdgeTX 2.8.0 or the current 2.9.0 (main branch) via cable on the serial port (ext. device) with the HorusX10S Express, this process takes well over 5 minutes. The same process with OTX 2.3.14 takes only a few seconds. Why does it take so much longer to flash external hardware? Is there a way to set the baud rate for the transfer? Or is it something else?
On the receiver you can observe during the flashing that the LED is red for a long time, then green, then there is a very short light green pulse, then it continues with red again and so it repeats. If I flash with OpenTx, then you only see flickering between light green and normal green.

### Expected Behavior

Rx Flashing via ext. port should be the same speed as with OpenTx

### Steps To Reproduce

flash an ArcherRx via Ext. Port and observe the LED on Rx and a watch

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

2.9.0 nightly and 2.8.0 release

raphaelcoeffic, 2022-12-25:
We would need to check if this is something with the baudrate, not quite sure right now. On the other hand, I believe this is the only possibility, as the receiver itself sends the requests for more data when it is ready to do so. So, apart from the baudrate, the receiver decides on the pace by itself.

The only way we could detect the issue would be to use a logic analyser while flashing the receiver with EdgeTX 2.8 and with OpenTX 2.3.14 and check the differences. We can then adapt for the differences.

3djc, 2022-12-25:
Don't think there is any difference, and it cannot take a couple of seconds with otx unless it fails somehow

ParkerEde, 2022-12-25:
I have indeed mixed something up there. I just compared it again directly. The Archer receivers seem to behave causally different than those from the X series.  
Therefore, it can be closed here.

closed 2022-12-25 by ParkerEde