EdgeTX/edgetx #2943: Horus X21S 6 position switch numbers freezing again since nightly update change.

Kevltan, 2022-12-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Moving the selector knob sometimes fails to index the numbers on the screen. Next click jumps over the missed digit.

### Expected Behavior

To index numbers on screen reletive to knob position. Was fine in 2.8.0 after firmware was changed due to a similar problem.

### Steps To Reproduce

As above.

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X12

### Anything else?

2.9 nightly selfbuild used.
Calibration does not solve the issue.

eshifri, 2022-12-26:
Interestingly enough, calibration in the TX16S simulator is broken completely: the 6POS is stack at 2 in the calibration screen and at 1 in the main screen.

philmoz, 2022-12-26:
@eshifri In the simulator calibration try setting the 6POS switch to position 4 when asked to center all controls.

eshifri, 2022-12-26:
@philmoz Now it is stack in pos 4 in both screens. :-)

philmoz, 2022-12-26:
@eshifri Very strange. When I calibrate it I move the 6POS knob slowly through each position a few times. I set it to 4 on the center screen then on the move screen I go 3, 2, 1, 2, 3, 4, 5, 6, 5, 4, 3, 2, 1, 2, 3, 4, 5, 6 and it calibrates fine. I avoid jumping randomly between values and also avoid jumping directly from 0 to 6 or 6 to 0 (which the knob control allows). Using the scroll wheel on the mouse seems the easiest method for me.

eshifri, 2022-12-26:
Thank you! I got it working after rebuilding simulator and companion in addition to libsimulator.

Kevltan, 2022-12-26:
Have you updated the nightly as the same problem still exists? For instance
if it gets stuck on 6 I have to go to 4 and then return to 6 before I can
get 5 position. This is not always true and can do it between any numbers.
Still works in 2.8 consistently.

eshifri, 2022-12-26:
@Kevltan It was my problem only: I was using old cpn and simulator builds. Nightly should be up-to-date.
An this was a simulator only problem. I thought it might have been related to your observation - looks like it was not the case.

Kevltan, 2022-12-26:
Yes it is definately related to the 2.9 nightly build as have again gone back to 2.8.0 and it works flawlessly. Something must have been changed or is in someway related to other recently edited code/parameters that the switch encoder uses. I have tried the starting at 4 in calibration as specified earlier @philmoz but this also does not solve the issue. Its almost like the encoded position tolerance has been changed or is not equal between the selected 1 - 6 detents.

eshifri, 2022-12-26:
@Kevltan I have a suspicion, but I do not have Horus.
So,
1. Do you have any way to check whether this is a display issue only, or the position is also incorrect?
2. Would you be able to test if I create a branch?

Kevltan, 2022-12-26:
Radio output positions correct. 
100%, 60%, 20%, -20%, -60%, -100%.
It is on the display itself bottom middle above 6 position switch knob. The 2.8.0 version had only lines with the selected number highlighted on its respective line. 2.9 has all the numbers 1- 6 instead of lines and number active is highlighted.

Kevltan, 2022-12-26:
will edit above and retry

eshifri, 2022-12-26:
I think I was able to reproduce it on TX16S (to some extent) and I have submitted the PR #2947  .
It will be compiled shortly. Would be great if you can give it a try on Horus.

Kevltan, 2022-12-26:
Absolutely solved the probem, superb @eshifri cracking piece of work 👌👍😁
Can you sort the IMU problem logged earlier in bugs #2786? 🤔😉

PR #2947 (2023-01-01): Fix display of 6POS display. (#2943)

Fixes #2943

Summary of changes: fix mismatch in value definition between checkEvents() and paint(0 for MainView6POS.
closed 2023-01-01 by pfeerick