EdgeTX/edgetx #2948: Radiomaster TX16S parameters for standard firmware on releases

Luznatural, 2022-12-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

More and more Radiomaster TX16S owners are implementing the internal bluetooth mod in their radios.
Every standar release of the firmware for this radio, that EdgeTx team release, has this option deactivated by default.
This fact oblige the users to have to produce custom firmware for their radios, this time with the bluetooth option ON. And not everybody is able to self create the custom firmware.
I have tried the firmware with the option ON in radios without the mod, and does not have any negative effect on them, as the internal bluetooth cannot be activated as it is not there.
Would it be possible to set bluetooth ON as default in all new releases that EdgeTX produce as default?

### Describe the solution you'd like

Every standard default firmware release that EdgeTX team produces for Radiomaster TX16S should include the option Bluetooth ON as default.

arthurson, 2022-12-27:
if buddy has option what extra function and language we want to choose, it will be even much better

pfeerick, 2022-12-28:
That is not a solution, as enabling it by default will result in the complete loss of one of the two AUX ports, regardless of if the user wants/has bluetooth or not. What really needs to happen is, like with some of the other options, for the driver port to be complete, thus it will then be configurable at runtime.

Luznatural, 2022-12-28:
That's a good interim solution and for sure will solve the problem for many.
Regarding making the driver port complete to have both AUX ports functional at the same time with bluetooth internal module, the solution is in our side?, or is in the side of ETX Dev Team? Is there a way that we might help, like pressing the company to find a solution, or similar?
Thanks for your efforts to make our life happier...

pfeerick, 2022-12-28:
It's an ETX problem, and a lot of the groundwork for all the other drivers/peripherals has been done, so hopefully, this will be something that will make 2.9, otherwise, it definitely should make the following release.

Luznatural, 2022-12-28:
Fantastic then Peter,
You have a solution in your mind.
Do you need any action from my side (like producing a new issue) or you will include it automatically on ETX 2.9?
Thanks again for your excellent support

rotorman, 2022-12-28:
Here a draft PR by @dlktdr working towards making Bluetooth an option to be selected for the serial ports: https://github.com/EdgeTX/edgetx/pull/2790

dlktdr, 2022-12-28:
It works with the TX16s as is (still needs some GUI love if not selected).  B&W currently the hold up especially the X9E with it's odd Bluetooth module. I know this module isn't compatible with much so I have dropped it in the most recent commits. Will update that draft now.