EdgeTX/edgetx #2950: X9E: copying models from TX16S to X9E via companion ends up in reboot

Gerold68, 2022-12-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have copied 6 models from TX16S via companion to the X9E.
If I then switch from the existing model (created on the radio) to a copied one, the radio reboots and the old existing model is still active.
 

### Expected Behavior

The copied models should work like on the source radio.

### Steps To Reproduce

1. read the setting from X9E radio
2. read the setting from TX16S radio
3. move one model with drag an drop between the two windows
4. maybe rearrange model picture, switches, used tx module 
5. write setting to X9E
6. disconnect the usb wire
7. select one of the copied models
8. then the radio reboots

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X9E / X9E Hall

Eldenroot, 2022-12-27:
I am not sure but you should export/import models only between same type of radio...

Gerold68, 2022-12-27:
Not really :-) Companion has the option to create radio profiles and if you switch between profiles, companion tries to converts the models and shows switches and knobs which have to setup manually.
By the way, the transfer between TX16S and Zorro works. It seems to be a X9E issue.

closed 2023-09-05 by raphaelcoeffic