EdgeTX/edgetx #2958: Lua functions should avoid setting values outside limits

jfrickmann, 2022-12-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On a number of `model.set*` functions, there are no checks on the values, so the user can set values that are invalid and potentially dangerous from Lua. As an example, a user can set the min. output limit to +3 from Lua, but it cannot be set to a positive value from the GUI. And apparently, the channel output goes to +100% when min. is set to +3 😦 
So I have started a branch where I want to go through the `model.set*` functions and add  limit checks, starting with `model.setOutput`.

### Expected Behavior

Lua should check values.

### Steps To Reproduce

See above.

### Version

2.4.0

### Transmitter

Other (Please specify below)

### Anything else?

No

TramALot, 2022-12-29:
I'm not a Dev but have you tried the newest version I think it's up to 2.8?

jfrickmann, 2022-12-29:
I don't think that this code was modified since it was carried over from OpenTX, but you can always look at t he blame, if you are curious.

gagarinlg, 2022-12-29:
There needs to be a central part of the software that provides the limits. This can then also be used from the UI and companion.

jfrickmann, 2022-12-30:
In principle, I agree 100% with that statement! Right now, however, it is implemented ad hoc in every version of the UI, and that is where I look for guidance.
Or we could just implement it all in Lua, and code all GUIs in Lua?!?

gagarinlg, 2022-12-30:
I am against doing all UI in lua, because I personaly do not like lua, but if it would be the best solution, ok...

The central limit provider can be created later, with your change at least the checks are done outside of tge lua scripts, so they are applied to all scripts, automatically.

jfrickmann, 2022-12-30:
Just be aware that it is not always as straightforward as giving a range of values. E.g. for logical switches, the allowed values for V1, V2 and V3 depend on the family of logical switch. And V3 sometimes depends on the value of V2. So it is really messy!

jfrickmann, 2023-02-27:
I have been down a rabbit hole of reading GUI code and trying to guess what the limits are for the various values in the many different cases. And it's going to be a lot of work testing that all of that works. Since I can see that people are now discussing a rewrite of the Lua API, I will quietly close this one ...

closed 2023-02-27 by jfrickmann

gagarinlg, 2023-02-27:
Not so fast, it is not going to be a complete rewrite.
We will clean up the mess, that we inherited from OpenTX, add UI widgets and (maybe) even create an app format.

lshems, 2023-03-01:
There are several ways of looking at it. 
Actually, a lua api call for a "set" should call the same function that the GUI "calls" for a set command. The validation should come from the called function.

For integer valued numbers you could make that function 'roll over' if it exceeds the upper bound. So upper bound is 200, lower bound is 100. Set to 250 and it is simply set to value (250 - lower bound) modulo (upper bound - lower bound) + lower bound. 
The function should 'gard' the limits, and should return the value actually set.
You could also have an option 'no-rollover' and an option 'force set' or soimething, that will either just give the upper bound if you set to a value exceeding, and lower bound if below lower limit. 'Force set' would only set if it is within limits.

In this way, you need a mimimum of overhead to start with, and could implement almost all "increase-decrease" scenarios with small step sizes (1, 5, 10?) depending on the range.

This would in my humble opinion be an extra function that is NON intrusive to both LUA and GUI, but would prevent wrong usage of lua set API's.

I am really no C++ programmer, so perhaps I'm uttering nonsense.

lshems, 2023-03-01:
You could even set a flag to check the limits. just return le the upper limit, or the lower limit, or the value that would have been set if the 'check' flag was not set.
One function call, with some flags, would do it all.