EdgeTX/edgetx #2960: missing LUA Values for TIMER

ParkerEde, 2022-12-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have read out the table for the TIMER in LUA. I found out that the parameters for "switch" and for the "countdown value" are missing. Furthermore I noticed that the parameters for "value" and "start" seem to be double.

### Expected Behavior

All timer parameters should be able to be set by LUA

### Steps To Reproduce

print the timer table und validate the settings with Radio

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

all previous versions of edgeTx are affected

philmoz, 2022-12-31:
It should be easy enough to add the missing values.

For the 'countdown start' value, the raw values for this in the code are -2, -1, 0 and 1 (corresponding to 5s, 10s, 20s and 30s). What value should be returned in the Lua table?
- raw value (-2 .. 1)
- adjusted raw value (0 .. 3)
- # of seconds (5, 10, 20, 30)

The switch value is also going to be a magic number and I'm not sure how best to present this to a Lua table.

Lastly 'value' = 'start' when the switch is reset - as soon as the timer starts running 'value' should change.

ParkerEde, 2022-12-31:
>Lastly 'value' = 'start' when the switch is reset - as soon as the timer starts running 'value' should change.
ah okay. thank you 

>The switch value is also going to be a magic number and I'm not sure how best to present this to a Lua table.
it would make sense here to implement it exactly as in the other tables, where switches can also be set. E.g. with the mixer, logical switches or special functions etc.? What do you think about that?

PR #2979 (2023-01-08): fix(lua) - Add missing switch and countdownStart properties to timer functions
Fixes #2960 

Summary of changes:

Add the switch and countdownStart properties to the getTimer and setTimer Lua functions.
closed 2023-01-08 by pfeerick