EdgeTX/edgetx #2961: full screen widget lost 13% area to spaces

offer-shmuely, 2022-12-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

the full screen widget have unneeded spaces around it, 
when they are not needed
I think it is the decision of the widget developer if a spaces are needed, and how match of the them

[image]

### Expected Behavior

as it was in 2.7
the widget should be allow to consume 100% of the available screen (480x227)

[image]

### Steps To Reproduce

create a widget that need the 480x227 pixels and try to display it without cuts

### Version

2.8.0

### Transmitter

FrSky X10 / X10S (ACCST), FrSky X10 Express / X10S Express (ACCESS), FrSky X12, Jumper T18, Radiomaster TX16S / TX16SMK2

offer-shmuely, 2022-12-30:
this calculation show a 13% of screen space lost for spaces

[image]

philmoz, 2022-12-30:
I suspect the problem is in line 122 of layout_factory_impl.cpp
      if (decorationSettings & 0x7) {
should be
      if (decorationSettings & 0xE) {

PR #2962 (2023-01-01): fix(ui) - Fix layout size issue (#2961) and improve code readability.
Fixes #2961 

Summary of changes:

Add enum for the 'decorationSettings' bits to make the code more understandable.
Fix the size check in 'getMainZone'.
closed 2023-01-01 by pfeerick