EdgeTX/edgetx #2963: NV14 REBOOTING  FTER TUNING RADIO OFF

dnielmtzfpv, 2022-12-31:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

i tried all firmware versiions  and  same resul i fflashed  using  companion and  i tried again using tx buddy and i tried  with firmware flasher  app too  and same ressult  my radio work  finei until i need to turnit off  it turns off but   1 second later it turns on  by itself

### Describe the solution you'd like

a way to turn it off wiout have to take of  batteries 

### Describe alternatives you've considered

none  i tried  to much times  and no solution

richardclli, 2022-12-31:
Did you plugged in the USB?  If you plugged in the USB, it will not power off but restart and enter the charging mode.

mattiachiarelli, 2023-01-16:
I have 2 nv14, one is fine after upgrading to 2.8 From 2.7.1

In the second one i upgraded directly from opentx and it is now impossible to turn off (so same sinthom)

Is there any test i can do to help figure out the bug?

gagarinlg, 2023-01-16:
You could open the radios and check the mainboard revisions

mattiachiarelli, 2023-01-16:
> You could open the radios and check the mainboard revisions

Done! They are both V1.6

gagarinlg, 2023-01-16:
Hmm, strange.
Did you try without the sd card and without the strap holder?

mattiachiarelli, 2023-01-17:
> Hmm, strange.
> Did you try without the sd card and without the strap holder? 

I've tried with the board out of the radio too because I thought it was the shell pressing the buttons

The bottoms work correctly because I can turn off the radio (but than it reboots itself)