EdgeTX/edgetx #2964: Global Variables display whole numbers when 0.0 precision is selected.

phileville, 2022-12-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When configuring a global variable, if you select for the precision field, the 0.0 option (to include decimal values), when you adjust the value for FM0 it still shows non-decimal numbers in the input field.  However, at the top you it shows the correct global variable value.   For example, it will if you set it to 20, the actual value will be 2.0.  This behavior exists in 2.8 as well as the nightly from 30.01.2022.  Also, in the nightly, FM0 is now labled as "mode1".. I am not sure why that is... in 2.8 it was labeled FM0.

### Expected Behavior

The correct numnber value should be shown based on the selected precision.

### Steps To Reproduce

Configue a global variable
Set precision to 0.0
Adjust values for FM0 - notice that the values will be in whol numbers.  

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

gagarinlg, 2022-12-31:
My gvar page PR fixes this

Eldenroot, 2022-12-31:
https://github.com/EdgeTX/edgetx/pull/2945

closed 2023-01-15 by pfeerick