EdgeTX/edgetx #2967: "Play Sound" special function does not play at startup

phileville, 2022-12-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The "Play Sound" special function does not play at startup even if the designated switch for the SF is in the position to activate it and regardless of the "repeat" setting.

The "Play Track" special function does play the sound at startup according to the repeat setting (!1x = no, 1x = yes).

Also for the label "value", I would recommend to change that to something more descriptive, like maybe "sound".

### Expected Behavior

The "Play Sound" Special Function plays the sound at startup according to the repeat setting (!1x = no, 1x = yes) like the "Play Track function"

### Steps To Reproduce

Configure special function "Play Sound"
Set **switch** to your choosing
Set **value** to anything you like
set **repeat** to 1x
turn radio volume up so that you can hear beeps
Set the switch to the postion to active SF - notice it beeps
Turn radio off and then back on - notice no beeps at startup.

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

mha1, 2023-05-06:
I can confirm the problem to be also present in 2.9

philmoz, 2023-05-06:
In the evalFunctions() method (radio/src/functions.cpp) the 'PlaySound' case checks for the audio queue being empty and does not play unless it is empty. The 'Play Track', and 'Play Value' cases do not do this.

On startup the queue is not empty as the startup sound and model name are queued first, so the 'Play Sound' SF gets ignored.

Removing the test for the empty queue solves this and makes it play on my TX16S.

Is there some history or reason why 'Play Sound' does this check on the audio queue?

pfeerick, 2023-05-07:
I don't know the reason, but it was originally done that way about 9 years ago back in OpenTX 2.2 and has only had a minor change (`empty` => `isEmpty`).

philmoz, 2023-05-07:
The only reason I can think of to have this check is so there is no delay between the SF going active and the sound being played.

Personally I would remove the test and have the sounds always play - even if delayed.

pfeerick, 2023-05-07:
Oh, interesting... when you go down to line349 it's not in that version (and never has been) - which doesn't really apply to us anymore since SDCARD should always be true atm. So yeah, let's remove that test, and see if are any side effects/complaints down the track 😆

mha1, 2023-05-07:
I actually looked at this yesterday too and built firmware without checking for empty queue. It works fine. And I dont think this was to prevent delays. Because if a sound is triggered and the queue is not empty there is no retry. The sound is lost.

I also think this check must go because this check dumps any Play Sound at any time (not only at startup) if the audio queue is not empty, e.g. LUA script or other SF/GF triggered Play Track just before Play Sound is triggered regardless of 1x or repetition. For repetition this means you might randomly loose Play Sound events. For single sounds you might loose the completely as in this issue.

PR #3567 (2023-05-15): fix(radio) - Fix SF 'Play Sound' not working on startup.
Fixes #2967 

Remove the check on the audio queue state from the 'Play Sound' code.
closed 2023-05-15 by pfeerick