EdgeTX/edgetx #2970: Proposed Curve TAB (curve selection)

JimB40, 2023-01-01:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Nope

### Describe the solution you'd like

Using 160px wide modules

Curve 1 selected
- Pressing ENTER or lonf tap brings conext-menu with (Edit, Preset, Clear) 

[image]

Curve 2 selected
- Pressing ENTER or lonf tap brings conext-menu with (Edit, Preset Clear) 

[image]

Empty curve 3 selected
- Pressing ENTER or lonf tap brings conext-menu with (Edit, Preset) 
- All 32 curves slots are visible 
 
[image]

Option
- Only slots with curve set are visible (ilke in Inputs)
- Presing ENTER or taping on + tile will create curve in first empty slot.

[image]

Eldenroot, 2023-01-01:
Nice, so if all will go in right direction we will have totally revamped GUI for curves!!! Nice, keep.good.job

philmoz, 2023-01-02:
The current curve list shows the curve 'Smooth' state. Can you make room for this in the new design?

[image]

philmoz, 2023-01-02:
What is the reason why a long tap is needed to bring up the context menu?

In the current version only a single/short tap is needed.

JimB40, 2023-01-02:
> The current curve list shows the curve 'Smooth' state. Can you make room for this in the new design?
 
Decluttering UI. This information is not needed as you see on preview If it's smoothed. If someone will really need we can add small icon instead of text that needs UI space and traslation.

> What is the reason why a long tap is needed to bring up the context menu?

My mistake. Short tap brings context menu.

Eldenroot, 2023-01-02:
I agree, smooth type is not necessary to display, maybe the icon idea is nice (somewhere over the curve, some symbol...).

philmoz, 2023-01-02:
This was relatively straightforward to implement; but my code changes are dependent on PR #2973 so I will post a PR when that gets merged. I went for the second option of only showing active curves and the '+' button to add a new curve.

Eldenroot, 2023-01-02:
Any screenshots to discuss? :)

philmoz, 2023-01-02:
[image]

[image]

[image]

JimB40, 2023-01-03:
I wonder why "Mirror" is in pop-up menu. 
Logicaly it should belong to Curve editing screen.

Instead of "Mirror" I'd rather see useful "Duplicate" feature.

offer-shmuely, 2023-01-03:
I agree that mirror not needed in menu
I also suggest that the number of points will be an icon as well, like the budge on smartphones
This design is missing the number of graph, like "cv1:thr" and not just "thr"

So the bottom space can now be left for description ;-) 
I keep forgetting what is the meaning of "kea" (knife edge adjustment)

Eldenroot, 2023-01-03:
> Remarks about pop-up menu
> 
> 
> I wonder why "Mirror" is in pop-up menu. Logicaly it should belong to Curve editing screen. This screen is to preview and manage curves so pop-up menu should hold Add (New), Duplicate, Delete , Edit, etc)
> 
> Instead of "Mirror" I'd rather ther see useful "Duplicate" feature.
> 
> 
> We should unify naming for removing data (that is also about other scrrens like Inputs, Mixes etc. a. Delete - it removes all data and frees slot ie. Curve, Input, Mix is not longer visible. b. Clear - IMHO "Reset" is better - it sets all Curve, Input, Mix data to default values but item is still visible.

Agree with all your points!

Eldenroot, 2023-01-03:
> I agree that mirror not needed in menu I also suggest that the number of points will be an icon as well, like the budge on smartphones This design is missing the number of graph, like "cv1:thr" and [...]

Actually it could be added together - number of points and smooth style - both icons.

About deacriptions - nice idea!

Yes, number of curve is needed.

Everything else seems to be really nice and looks perfect! Good job

philmoz, 2023-01-03:
Some good ideas, although some will need to wait for a while.

- Show curve number and name in title - added
- Remove mirror - there is no mirror function in the curve editor so removing it here would lose the feature altogether.
- Badge icon for # of points - I don't really like this. Badges (on phones) indicate some new/additional information is available which would not be the case here.
- Smooth icon - future enhancement
- Description - future enhancement

JimB40, 2023-01-04:
Displaying curve slot ID (label 'CV' plus slot number) is not really needed. Later in OS when you choose curve as parameter source either curve slot ID or curve user defined name is displayed. Not both.

Sometime ago @raphaelcoeffic introduced Inputs & Mixes list manadgement hiding empty slots.
While this has advantage of showing only relevant data it has also few overlooked drawbacks with current UI.
1. Add button is always last
2. Automated curve naming can have holes in numbering.
- Add 9 curves it wil show CV1,CV2,CV3,CV4,CV5,CV6,CV7,CV8,CV9.  
- Delete CV3 & CV6 and list wil show CV1,CV2,CV4,CV5,CV7,CV8,CV9.
3. New curve not always showing as last one.
- In above example add new curve and it show as CV3 (first empty slot). If there are let's say 40 items it may be tricky to find 'Where is my new curve?'

To recap new implementation of item's list management is now somwhere between using numeric ID and not using numeric ID :)

philmoz, 2023-01-04:
The way I have implemented it, any new curve added is automatically selected and the page is scrolled to show it.

pfeerick, 2023-01-15:
Just commenting here to keep the discussion on this together rather than overflowing to the implementation PR...

> This design is missing the number of graph, like "cv1:thr" and not just "thr"

Actually, no - that is not there on the current UI implementation, not is it strictly needed as JimB40 commented... CV# is only used until you specify a name... at which point that is used instead. 

> Badge icon for # of points 

I agree... I don't like this either for this purpose... this is a data element, not the number of notifications, etc. Icons for Standard vs Custom, and Smooth vs non-smoothed, yes, but not that.

offer-shmuely, 2023-01-15:
The view is probably implemented depended, with a view to the future.
if the implementation is add/move/delete as suggested to b in the future, then indeed, no need for C2:abc
if the implementation is actually slotted, i.e. you have 16 slot, and the key is the slot id, then to hide it does not serve the user 

please note that in option #1, if 2 graph have the name FLP vs FLP, it will create a mess, so we should protect against saving two graph with same name.
on option #2, two similar names, does not create a problem, since the key is the slot-id, and it just more clear CV2:FLP & CV8:FLP, and since the view we see only 3 graph, (8 on old views) double naming can be become more common.

pfeerick, 2023-01-17:
Oh, indeed... I think that using short names and having the number helps prevent confusion, and it does not harm to show both if the UI can easily accommodate it without making it cluttered and overloaded. I was more commenting on the fact that it is not missing with respect to 2.8.0 UI (or earlier IIRC). 

This is where we come back to needing short and long-form names / short name + label/description/summary/whatever (e.g. "KEA" - knife edge adjustment). i.e. the choice field might own show "KEA", but when you click on it, you get "KEA - Knife Edge A..." in the list of choices (if it were say limited to 12 characters), and the full name when you are in the curve edit screen.

offer-shmuely, 2023-01-17:
Regardless if the title have the CV1 or not (not that important)
I like to suggest two things:
1. to space it down so at least 4 graph can be on the same screen (will be very helpful for power users)
2. although I usually like the overview very match, in this case I do not see any real info to display the number of point nor the smooth/standard, once we see the graph itself. 
on the other I would keep the space that now vacant, for the future description (hope it will happen) 

Here is a sample:

[image]

Eldenroot, 2023-01-17:
Good feedback – personally, I prefer to have “CVx” displayed.

I agree that we do not need to see how many points the curve has, it is pointless. Also, I was thinking about “smooth” - I think it is not important to display on overview at all. So it could save a lot of space. I really like the ability to add a description.

4 curves/page – yeah, why not – there is a huge number of users which do not like/do not want to use touch UI, so this is a good way :)

offer-shmuely, 2023-01-18:
another option is to add the points, this have a real additional value
something like the companion do, (but we need to more creatively on graph with 19 points)

[image]

closed 2023-01-23 by pfeerick