EdgeTX/edgetx #2974: Wiki build targets companion28 and simulator28 do not exist

adelton, 2023-01-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

The https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Ubuntu-20.04#building-companion-simulator-and-radio-simulator-libraries suggests that after `make native-configure`, commands
```
make -j`nproc` companion28
```
and
```
make -j`nproc` libsimulator && make -j`nproc` simulator28
```
should work. However, `make` reports
```
make: *** No rule to make target 'companion28'.  Stop.
```
and
```
make: *** No rule to make target 'simulator28'.  Stop.
```
respectively.

### Expected Behavior

Following https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Ubuntu-20.04 should not yield any errors.

### Steps To Reproduce

1. Follow the steps on https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Ubuntu-20.04 in an attempt to build the companion and simulator.

### Version

2.8.0

### Transmitter

Other (Please specify below)

pfeerick, 2023-01-02:
Are you actually trying to build the 2.8 branch, or `main` (thus currently 2.9,  and `companion29/simulator29` - the wiki could explain that better rather than having "magic numbers" in the examples). 

If you're trying to build the 2.8 branch, and getting that error, it would suggest there is something wrong in the build environment.

rotorman, 2023-01-02:
Just updated https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Ubuntu-20.04#building-companion-simulator-and-radio-simulator-libraries to better reflect this.

adelton, 2023-01-02:
Yes, I'm using the 2.8 branch, specifically the v2.8.0 tag (release).

I just reproduced the problem from scratch, starting with a fresh Ubuntu 20.04 VM from https://cloud-images.ubuntu.com/releases/focal/release/ubuntu-20.04-server-cloudimg-amd64.img, and then I ssh'ed to it and did just the following steps:
```
root@edgetx-ubuntu-20-04:~/edgetx-2.8/build-output# history
    1  apt update
    2  apt upgrade
    3  wget https://raw.githubusercontent.com/EdgeTX/edgetx/main/tools/setup_buildenv_ubuntu20.04.sh
    4  chmod a+x setup_buildenv_ubuntu20.04.sh
    5  ./setup_buildenv_ubuntu20.04.sh
    6  source ~/.bashrc
    7  git clone --recursive -b v2.8.0 https://github.com/EdgeTX/edgetx.git edgetx-2.8
    8  cd edgetx-2.8 && mkdir build-output && cd build-output
    9  cmake -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DLUA_MIXER=YES -DCMAKE_BUILD_TYPE=Debug ../
   10  make native-configure
   11  make companion28
```

And I get
```
make: *** No rule to make target 'companion28'.  Stop.
```

When I use the target without that version string, it works:
```
root@edgetx-ubuntu-20-04:~/edgetx-2.8/build-output# make companion
[100%] Built target native-configure
Scanning dependencies of target companion
Scanning dependencies of target yaml-cpp
[  0%] Building CXX object companion/src/thirdparty/yaml-cpp/CMakeFiles/yaml-cpp.dir/src/binary.cpp.o
[  0%] Building CXX object companion/src/thirdparty/yaml-cpp/CMakeFiles/yaml-cpp.dir/src/convert.cpp.o
In file included from /root/edgetx-2.8/companion/src/thirdparty/yaml-cpp/include/yaml-cpp/node/convert.h:18,
                 from /root/edgetx-2.8/companion/src/thirdparty/yaml-cpp/src/convert.cpp:3:
/root/edgetx-2.8/companion/src/thirdparty/yaml-cpp/include/yaml-cpp/node/impl.h: In member function ‘YAML::Node& YAML::Node::operator=(const T&)’:
/root/edgetx-2.8/companion/src/thirdparty/yaml-cpp/include/yaml-cpp/node/impl.h:212:11: warning: ‘operator=’ should return a reference to ‘*this’ [-Weffc++]
  212 |   return *this;
      |           ^~~~
[  0%] Building CXX object companion/src/thirdparty/yaml-cpp/CMakeFiles/yaml-cpp.dir/src/directives.cpp.o
[  0%] Building CXX object companion/src/thirdparty/yaml-cpp/CMakeFiles/yaml-cpp.dir/src/emit.cpp.o
[  0%] Building CXX object companion/src/thirdparty/yaml-cpp/CMakeFiles/yaml-cpp.dir/src/emitfromevents.cpp.o
[  0%] Building CXX object companion/src/thirdparty/yaml-cpp/CMakeFiles/yaml-cpp.dir/src/emitter.cpp.o
[...]
```

rotorman, 2023-01-02:
Well spotted. I just re-edited https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Ubuntu-20.04 to match this.

adelton, 2023-01-02:
Thanks for the confirmation and for the fix.

pfeerick, 2023-01-02:
Ah, yes, sorry, just before bed is a bad time to read issues 😆 . The actual build/make command doesn't change - it's always companion, simulator, lib-simulator, firmware, etc, but the resulting executable for companion and simulator will have the version number in it (on Linux, and presumably on Mac also - on windows it is just `companion.exe`).

closed 2023-01-02 by pfeerick