EdgeTX/edgetx #2975: Horus X12s touchscreen inverted

Unb0rn, 2023-01-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

It looks like there are some issues with my modded X12s (externalaccessmod, ISRM, touchscreen mod)
After installing the current (v2.8.0) release I found that my touchscreen is inverted for some reason.
Last time I checked (around [those](https://github.com/EdgeTX/edgetx/discussions/2326) commits) everything worked fine and I can confirm that these builds still seem to work correctly.
I also found an issue #2063 which suggests re-setting touch panel config. However, it looks like VCP is not responding at all Tried both on Windows and on Linux, the `ttyACM0`  or `COM4` appear (depending on OS, of course), I can connect to it (tried on 9600, 115200 and 400000 bauds) and see the characters typed in, but pressing enter does nothing.
`USB-VCP` is set to `CLI`, of course.
There is also another issue #2837 for another transmitter, could it be related?

### Expected Behavior

The touchscreen works fine and aligned correctly. VCP CLI responds to requests

### Steps To Reproduce

1. Install v.2.8.0
2. Add Touchscreen
3. Test it
4. Try connecting via VCP

### Version

2.8.0

### Transmitter

FrSky X12

gagarinlg, 2023-01-02:
See
https://github.com/EdgeTX/edgetx/issues/2063#issuecomment-1257155877

closed 2023-01-02 by gagarinlg

Unb0rn, 2023-01-02:
@gagarinlg Yep. I mentioned it. The problem is - VCP is not functioning. I'm not sure if the problem is related to my mods, to my device or in any way related to my setup or maybe it is a broader problem

rotorman, 2023-01-02:
Possibly a line termination character error - is CR+LF sent on Enter?
Can you try hTerm under Windows: https://www.der-hammer.info/pages/terminal.html and set CR+LF to be sent on enter.

Unb0rn, 2023-01-02:
Yup, that was the problem, thanks!
Shouldn't we revert from windows CR-LF style to standard CR? It was a bit confusing)

Luznatural, 2025-02-12:
Just for general information:

The easiest way to reset the inverted screen for X12S radios is using any Android tablet/phone.

In my case, I used a very simple and free application named "Serial USB Terminal" by Kai Morich, that you can download free from Play Store. I do not know if there is something similar for Apple phones/tablets.

Once Installed, you simply have to connect the radio to the tablet/phone, and select in the radio the opción VCP (serial).
As the program takes care automatically of the parameters of the connection, you do not need to worry about baud rates, COM ports, etc.

You simply touch the intro line, and the connection is established. Then write "reset_GT911" in the most lower line, press the arrow (intro)  and the screen shows the "reset to 0" response, and then the screen works perfectly.
Disconnect everything and you are ready to go.

Using windows for doing that is becoming more and more complicated, as very frequently Windows does not allow easy access to COM ports, as those ports are now being used for some other programs, and very frequently you do not get access to create correctly the COM port for the radio.

HinkePank, 2025-04-04:
Hi Luznatural,
thank you very much. I was totally desparate about getting no cwnnection to the CLI. All Windows 11 PC and Laptop could not start the interface ( was shown in device-manager as inactive ).
Failed also to get KNoppix to start on my PC's ( TPM ? ) .
Anyway, your idea worked at the end !
Now I got a real fancy X12S with a wonderful tochscreen and the nex Edgtx 2.10.6.
I bought the touchscreen directly from Radiomaster, could not get it anywhere else. It came with mode 109 - inverted. 
Thanks again
Harald

HinkePank, 2025-04-04:
Hi Luznatural,
I also have the Radiomaster tx16s with the internal miltimodule - my X12S also has the multimodule, so functionalitywise they are equal.
The tx16s is a bit smaller and the Lipos are easy to charge. I have Lipos in the X12S but not the Lipo-charge module ( nowhere to buy anymore ).
The position of the screen at the X12S is way better than on the Radiomaster. I like this very much.
The finger-potis are also better to control - I love the X12S and the EdgeTX is getting more perfect every version.
I can recommend this mod for every Horus-owner !
👍

Tots2Hots, 2025-05-14:
Just did this with the Serial USB Terminal app.  Mini USB cable to USBA to USBC adapter to the phone.  Few posts above it says use line "reset_GT911".  That isn't correct.  Its "reset_gt911" has to be lowercase for it to work.  No quotes when typing the line either.  

Finally got the old TX16S screen I cannibalized out of a broken radio working properly.