EdgeTX/edgetx #2981: ETX 2.8: Error message, freeze and EM when 64 mixer limit is reached

RC-SOAR, 2023-01-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

To reproduce, either in the Companion 2.8 sim, or on the transmitter running 2.8.0:

1. Activate an existing model setup
2. Go to the Mixers menu
3. Duplicate arbitrary mixers to bring the mixer count to 64 
4. Now click on any mix

An error message "No free mixer" is displayed. The tx freezes. Disconnect the battery. 
Reconnect the battery, the transmitter goes into EM.
Normal operation is restored by cycling the power

### Expected Behavior

No error should be raised.

### Steps To Reproduce

See above for steps to reproduce.

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

[image]

RC-SOAR, 2023-01-05:
I've tried again, with the latest nightly 2.9.0 (e1a80667), with the attached .yml file which has 64 mixers.

When I click on any mix, the tx appears to freeze but recovers in about 10 secs. It's better, but it's still not right.

Is there any reason to issue an alert just be clicking on a mix? In OpenTX, the alert is only be issued when the user is attempting to add the 65th mix (by choosing the 'Insert' option in the context menu).  This is IMO the way it should work.
[model1.yml.txt](https://github.com/EdgeTX/edgetx/files/10353108/model1.yml.txt)

philmoz, 2023-01-05:
My change to fix this is still pending and has not been merged into the build - it won't get fixed until that happens. 
I don't know why the behaviour would have changed, it crashes for me with the current branch.

The insert/paste menu options are not shown if there are no mixes available - I guess this is an attempt to tell the user why.
Perhaps it should only be shown once.

RC-SOAR, 2023-01-05:
Or perhaps re-instate the inserte/paste options, and just issue an error message if the 64 limit is about to be exceeded? In other words, display the error message when an error is actually being committed.
As it stands, the user is confronted with error messages, each of which must be dismissed, when simply inspecting existing mixes.

pfeerick, 2023-01-17:
The immediate fix is merged, but I do agree that the order is probably back to front... makes more sense to leave the options present, and you get the message if you try to use them.

PR #2985 (2023-01-17): fix (ui) - Fix crash on model inputs and model mixes pages if limit on mixes/expos reached.
Fixes #2981

Summary of changes:

The _run_popup_dialog function crashes on the model inputs and model mixes pages (I don't know why).

This change replaces the POPUP_WARNING call with a MessageDialog popup on the current window - it seems to achieve the same result without the crash.

It also uses the translated string for the warning popup title instead of a hard wired english string.
closed 2023-01-17 by pfeerick