EdgeTX/edgetx #2999: Locking up when scrolling through models

TPSDrummer, 2023-01-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Hold down the MDL button until the Select model screen appears. I have a label called HELI with all my helicopter models which is checked. I assume it remembers my last selected label.  I am in the HELI list. When using the jog dial to scroll down through the models the radio will freeze. None of the navigation buttons work and the power button doesn't respond. Changing toggles will play the track associated with them.  I have to pull the battery. Upon reboot it says EMERGENCY MODE. The power button works and I can shut down and restart. Then the radio works until I scroll through models again. 

The display shows 4 models per screen. I have 9 models under HELI and I can scroll through the first 6 (screen had to scroll one row) then when I have to scroll to #7 and the screen has to scroll a second row it always hesitates. Most of the time the hesitation becomes a lock up and I have to pull the battery.

### Expected Behavior

Scroll to the next row of models to select and not lock up.

### Steps To Reproduce

1. Power on normally
2. Hold MDL button until Model select screen appears
3. Scroll down using the jog dial past the 6th model image
4. Unit will hesitate and usually will freeze

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

gagarinlg, 2023-01-06:
Do you have images for the models and how large are those?

TPSDrummer, 2023-01-06:
Hello and thank you for the response. All models have images. List below.

gagarinlg, 2023-01-06:
Attachments don't go through mails into github issues

pfeerick, 2023-01-25:
@TPSDrummer Can you check the file size/image size of those images? If they are too big (or perhaps a corrupt image?) that could be the cause of the lockup.

TPSDrummer, 2023-01-25:
I deleted all the images and the issue seems to be fixed.

closed 2023-05-05 by gagarinlg