EdgeTX/edgetx #3000: Logical function comparison with VG use wrong scale

psychotik2k3, 2023-01-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have an input6 that does |x| to the elevator input so i have the delta of my elevator.
input7= S1 value weight 50% offset 50 so i have a 0-100 value as output

i have VG1 that i fill with the value of input7 in special function.

now if i create 2 logical switches:
L01: a>b input6 VG1
L02: a>b input6 input 7

if i center S1 both VG1 and input 7 got a 50 value
now if i move my elevator L01 will be true when input6 is at more than 5%
while L02 will be true when my elevator is more than 50% (or less than -50%)

if i turn S1 to max then L01 will be false until input6 is at 11%

i on't know why there is a 0.1 factor between the VG and the channel.
I trie to reduce the min/max values of VG1 to -100/100 instead of -1024/1024 when i discovered taht but that didn't change the problem only limited the values of VG1

### Expected Behavior

if VG1 is set to 50
a>b with a=|elevator| and b=VG1 shoul be true only when elevator is at either more than 50% or less than -50% 

### Steps To Reproduce

1 create an input6 with func |x| input elevator
2 create a logical function a>b with input6 and VG1 as inputs
3 set 50 to VG1
4 move elevator to +25%
5 L01 should be false (right now it is true when elevator is more than 6%)
6 move elevator to +60%
7 L01 should be true

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

[image]
[image]
[image]
[image]
[image]

eshifri, 2023-01-07:
Can you please provide your GV1 setup: units and precision.

psychotik2k3, 2023-01-07:
they are default.
i tried changin the unit to percent but it does the same
Here are the 2 settings i tried:
[image]
[image]

psychotik2k3, 2023-01-07:
if i want to have the logical switch to change at around 50% i have to set the VG1 to 500%

eshifri, 2023-01-07:
Thank you!
Can you please try setting precision to 0.0?