EdgeTX/edgetx #3001: Support for cheap common boards (Mega2560 / STM32F103C8T6)

TheMinimazer, 2023-01-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I wish to create a custom-shaped transmitter using off-the-shelf components, rather than just buying a pre-built transmitter and pulling it to pieces. OpenTX 2.1 was compatible with the Mega2560 chips and related boards (Arduino 2560 and clones), and it'd be nice to see more easily-accessible open-source firmware availability (and possibly ELRS availability) on cheap hardware. Other options require more know-how than simply buying a dev board and following some relatively simple diagrams and instructions such as what exists for OpenTX2.1 in [https://github.com/opentx/opentx/tree/2.2/doc/Mega2560](https://github.com/opentx/opentx/tree/2.2/doc/Mega2560) .

### Describe the solution you'd like

Add support for development boards such as Mega2560 (if possible), STM32F103C8T6 (aka blue pill), STM32F411 (aka black pill), or STM32F407VET6, such that they could be flashed from the companion like a pre-built transmitter.

### Describe alternatives you've considered

1. Deal with lack of recency/features with OpenTX 2.1
2. Compile OpenAVRc builds (I can't easily find a simple companion-style solution in OpenAVRc)
3. Build my own OpenTX 2.3 versions with [https://github.com/jemo07/opentx_STM32F407VET6](https://github.com/jemo07/opentx_STM32F407VET6) and working out pinouts for the board to connect pots, switches, etc.

raphaelcoeffic, 2023-01-10:
Atmega and F103 won't have enough resources for anything useful. The other 2 are already used in some targets. The question is: what kind of targets should these be? As we are moving towards unified targets, supporting these could be done easily, however it is quite out-of-focus for now.

TheMinimazer, 2023-01-10:
Not surprised re: AtMega and STM32F103 being weak.

What targets are the other two already used in?

I am not 100% sure on what the question "what kind of targets should these be?" means, but if it is a question of what the use-cases of these boards would be, I am interested in the use of a DIY solution to obtaining a land/surface (a.k.a. pistol grip) transmitter  for controlling complex (4+ channels) land vehicles. I guess there could also be use for these in creating more accessible/disability-friendly controllers, but that would not be what I am intending.

It is a shame there are no commercially-available surface transmitters with OpenTX/EdgeTX, and it'd be nice if one could be created easily (just plug in and upload firmware, rather than having to read code to find pinouts and then compile the code by oneself) with the ability to swap external modules (e.g. swapping between ELRS and 4in1 modules).

rotorman, 2023-01-10:
> What targets are the other two already used in?

Here a list of some microcontrollers known to me by the EdgeTX supported radios presently:
STM32F205VGT6
STM32F205ZET6
STM32F407VET6
STM32F429BIT6
STM32F439BIT6
First 3 are for black and white screen radios, the latter 2 are for color screen radios with beefier graphics and external SDRAM.

TheMinimazer, 2023-01-12:
I found that the STM32F407VET6 is used in the RadioMaster Tx12. So if that microcontroller is supported already, does that mean that I could:
1. Buy a [STM32F407VET6 development board with screen from AliExpress](www.aliexpress.com/item/4000953664266.html)
2. Use Zadig to make it bootloadable as per Page 13 of the [RadioMaster Tx12 Manual](https://manuals.plus/wp-content/sideloads/radiomaster-tx12-manual-optimized.pdf)
3. Flash it with the EdgeTx Companion, by selecting 'Radio Type' as 'Radiomaster Tx12'
4. Find the correct pin assignments for what components connect to which pins in one of the .h files in [EdgeTX/edgetx/tree/main/radio/src/targets/common/arm/stm32](https://github.com/EdgeTX/edgetx/tree/main/radio/src/targets/common/arm/stm32)
5. Build a DIY transmitter around it by using switches, potentiometers, etc. connected to the dev board

Would that work? And if so, which file should I be looking in for the pin assignments?

rotorman, 2023-01-12:
You could try to use a STM32F407VET6 Development Board, but would need to likely adjust the I/O pins in the code (assuming the external oscillator frequency matches, otherwise would need to adjust also the PLL values).
Present EdgeTX code for black and white targets with F407, can drive only 128x64 pixel or 212x64 pixel monochrome displays, not the color display included in that kit.
Instead of Zadig, best have a look at official ST tool - STM32CubeProgrammer https://www.st.com/en/development-tools/stm32cubeprog.html
Also consider getting yourself a Serial Wire Debug (SWD) capable in-circuit-debugger, such as ST-Link https://www.st.com/en/development-tools/st-link-v2.html or J-Link https://www.segger.com/products/debug-probes/j-link/models/j-link-edu-mini/ or similar. As, if you need to modify the code to get your changed I/O configuration running, it is easier to debug the changes via in-circuit-debugging.

gagarinlg, 2023-01-12:
If you wait a bit, all the I/O config will get much easier, when the switch-refactor branch is merged