EdgeTX/edgetx #3005: System locked when RTC data is broken

padcom, 2023-01-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When the RTC data is broken, the transmitter will not turn on. The bootloader starts ok. No indication of the situation of broken RTC data.

### Expected Behavior

The transmitter turns on.

### Steps To Reproduce

1. Get a transmitter with broken RTC data
2. Turn the transmitter on
3. Observe it is locked

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

The problem is with `__mktime_internal` and its processing:

```
  /* Repeatedly use the error to improve the guess.  */
  for (t = t1 = t2 = t0;
       (gt = guess_time_tm(year, yday, hour, min, sec, &t, ranged_convert(convert, &t, &tm)), t != gt);
       t1 = t2, t2 = t, t = gt) {
    if (t == t1 && t != t2)
      goto offset_found;
    else if (--remaining_probes == 0)
      return -1;
  }

 offset_found:
  *offset = guessed_offset + t - t0;
``` 

With broken data this loop never finishes.

raphaelcoeffic, 2023-01-10:
We should probably add some unit tests for `gmktime` to test this properly with invalid values.

padcom, 2023-01-10:
The idea behind this PR is to know when the problem occurs to help diagnose it - not to fix the underlying problem. It occurs rarely and at present finding out what happened is next to impossible for users without programming skills.

I presume there will be another PR, done by someone who has much deeper knowledge about the platform, that will try to address the root cause of the problem.

PR #3006 (2023-01-15): Added turning the red LED before RTC data and blue LED afterwards
With the LED turned RED if the RTC data retrieval fails it will stay red. This will at least be an indication that the RTC startup encountered a problem. For now there is no indication that this problem occurred.

It doesn't really fix #3005, but helps in diagnostics.

Summary of changes:

Added red LED before initializing RTC and blue LED afterward.
closed 2023-01-15 by pfeerick