EdgeTX/edgetx #3014: Name change in model's context menu

JimB40, 2023-01-11:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Feature allows to assign label not edit. Editing labels is different feature.
[image]

### Describe the solution you'd like

Change menu option from "Edit Labels" to "Assign Labels"

gagarinlg, 2023-01-18:
It is not only assign, but also unassign, so if you want to make a change, it might be "change labels", but not "assign labels"
Maybe the native english speakers could offer their opinion?

philmoz, 2023-01-18:
Label is a verb as well as a noun so perhaps using 'Label model' might work and is consistent with the options above it.

commit fbfea8db (2023-03-08): chore: Change "Edit Labels"
Resolves #3014
commit 09b677bd (2023-03-13): chore: Change "Edit Labels"
Resolves #3014
commit 83c0ff54 (2023-03-13): chore: Change "Edit Labels"
Resolves #3014
commit ac6c1447 (2023-03-13): chore: Change "Edit Labels"
Resolves #3014
closed 2023-03-23 by pfeerick