EdgeTX/edgetx #3015: Select Model - name change

JimB40, 2023-01-11:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

This screen now serves as set of features that allows different tasks (not only selecting)
Adding model, Selecting active model, Deleting model etc. 
So name doesn't reflect content.
[image]
[image]

### Describe the solution you'd like

Change name from "Select Model" to "Models Manager"

pfeerick, 2023-01-12:
"Models Manage", or "Manage Models"?

JimB40, 2023-01-12:
Yep all of them will fit. "Configurator" is too long.
If not "Models Manager" I prefer "Manage Models"

MicheleVilla, 2023-01-12:
the same question could be proposed for Radio Setting
It also contains tools like Spectrum analyzer and more that might deserve a separate section or a different name

pfeerick, 2023-01-12:
The Tools tab is going to be split out to its own icon/section at some point.

JimB40, 2023-01-12:
At last called "APPS" :)

gagarinlg, 2023-01-18:
Do we need to make a poll to decide if this change is going to happen?

JimB40, 2023-01-25:
@gagarinlg Don't think so. As much as I understand power of habits, generic name reflects better what's inside this module. 
Btw in computer apps what we see on topbar is "File' not 'Open file'.  In 3.0 UI there will be feature to create 'shortcuts' in quick menu so user will have possibilty to place 'Select model'. Still OS navigation tree names will remain generic.

closed 2023-03-23 by pfeerick