EdgeTX/edgetx #3017: Add collapsable model config settings in B&W radios

phileville, 2023-01-11:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The model setup page in B&W radios has too many configuration options and it is often hard to find the option that you want change.

### Describe the solution you'd like

I would like to have a **+** symbol added after the **Preflight Checks** option in model settings, that when selected, expands/makes visible all the configuration options which are: Checklist, T-Warning, Cust-Pos, Pos- %, S.Warning , Pot warn, Ctr Beep
Once the options are expanded / visible, the **+** turns to a **-** symbol, which when clicked will collapse/hide the options again. 

Additionally, I would like to add a **Throttle +** label in the configuration options that has the same expand/collapse capabilities for the Throttle Settings, which are: T-Reverse, T-Source, T-Trim-Idle, T-Trim-Sw 

These changes, combined with the changes in Issue #3016 will reduce the visible configuration options in model settings from 44 to 18.

philmoz, 2023-04-18:
I will look at adding this (and #3016) to PR #3484 as I'm already messing with the settings pages.

pfeerick, 2023-04-19:
An alternative to this if it makes it simpler for implementation would be to have a subpage with those settings. i.e. like with the joystick channel settings

PR #3484 (2023-05-22): feat(radio) - Add options to disable radio and model menu tabs.
Resolves #1692, resolves #3016, resolves #3017 and #1249

Summary of changes:
- Update radio and model data structs
- UI for color LCD radios
- Convert existing 'noGlobalFunctions' property.
- Language strings.

TODO:
- [x] B&W radio UI
- [x] Companion support

Notes:
- The existing 'noGlobalFunctions' model property is converted to the new model property and is reset. UI for this removed.

[image]
[image]

[image]
[image]