EdgeTX/edgetx #3018: Trim overide in Special Functions

Kevltan, 2023-01-11:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Trim overide in Special Functions

### Describe the solution you'd like

To be able to instantly overide trim setting with a switch or logical switch. 

### Describe alternatives you've considered

couldnt find one.

philmoz, 2023-01-17:
You can do this with flight modes; however you have to set it up in Companion, you can't do this on the radio at the moment.

Also note that the maximum value you can set for trim in Companion is 125, and min is - 125.
The actual range for trim on the radio is -128 to 128 so if you actually need full up or down trim you will need to manually edit the model YAML file and alter the trim settings.

You could also do this with a combination of logical switches, special functions and Lua scripts.
If you want to try doing it this way let me know and I will post details.

Kevltan, 2023-01-17:
Is it possible to create a 2 step trim in the firmware which can be
activated by either logical switches via Special Functions (trim override)
or the trim lever itself. It must show a visable move on the trim icon on
the screen as I use a RDT widget that shows trim position. Can the firmware
be edited by me to create this 2 step trim movement  instead of the coarse,
fine etc. and if so where can I find the code to edit, to do this? I think
this option would be a good implementation in EdgeTX as I believe it is an
option in some of the mainstream radios.