EdgeTX/edgetx #3024: Add missing Enable switch to SF/GF functions

JimB40, 2023-01-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Lack of Enable control in below stated Specia/Global Functions:
- Play Sound
- Play Track
- Play Value
- Lua Script
- BgMusic
- BgMusic II
- Vario
- Haptic
- SD Logs
- Screenshot
- RacingMode
- No Touch
- Set Main Screen

### Expected Behavior

Implement Enable control

### Steps To Reproduce

-

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

eshifri, 2023-01-13:
It is easy to do, but it means that pretty much every CF/GF needs to have "enable" control. 
The exceptions are RESERVED amd TEST functions.
Is it the intent?

JimB40, 2023-01-13:
I was trying to confirm with @pfeerick why some of SF/GF functions have this Enable feature and others don't.
Short answer is 'we don't know' :)

So putting heritage apart, I was thinking if this feature is valuable, or just another feature 'let's have it why not'
Most SF/GF are 'fixed' in my configs. Like Volume is on S1, Brightnes on S2, Screen switching on 6P switch and I don't change them (switch them on/off)

The only valuable case I was thinking of was to create 'scenarios'. So for example you have 3 different functions assigned to the same switch and you quickly toogle just to use one of them. But then ETX shoud rather group those SF/GF and allow to enable only one.

So to make UX consitent we should consider adding Enable to all or ... ditch Enable for all :)

But maybe there are other areas when this Enable for SF/GF is useful.
Any plane pilots over there using this feature? @jfrickmann , @lshems

PS.
For sure I'd change one thing. When new SF/GF is created default Enable should be ON as in my case everytime I have to switch it

lshems, 2023-01-13:
OK, why is the on/off switch useful:

because people do tend to have a habit of creating one master template that is used over and over again, and indeed, as you said, different versions with different settings for the same SF can then be in that template, and you can simply toggle between the ones you want to be active.

Also, you can of course inactivate a SF by deleting the trigger, but then the SF disappears completely, and I don't even know if the settings will be restored once you set a trigger again..

In my humble opinion, there are some solutions:
Create an OFF trigger, which allows to show SF which you do not want to have active, but keep settings from.
Allow the OFF trigger to be reverted to the last set trigger on that SF as one of the options in the trigger select list.

In this way, you have a similar or better functionality than we have now, that is applyable to all SF in general.

lshems, 2023-01-13:
OFF trigger is thus not the same as NO trigger at all.

pfeerick, 2023-01-13:
Basically they should exist for all SF/GF. It has nothing to do with plane pilots, and more to do with complex configs, organisation, and as you say "scenarios". Trainer Mode and Instant Trim are also examples of ones that you might want to be able to leave present, but toggle on and off as needed. 

Instead of messing with the "Switch" parameter (referred to as trigger above), I would stick with the enable/disable toggle. Which probably should be in the header bar, and yes, enabled by default. If it is in the disabled state, in the overview it should be visually different to one that is enabled (regardless of if that is active/inactive by virtual of the switch/trigger).

elecpower, 2023-01-13:
As a fixed wing pilot I use templates and have many SFs and a few GFs. Having the consistency to enable/disable every function type is something I have been waiting for. So much better than deleting and re-adding those without enable/disable flag.

Some of my functions are used to provide audio information when testing new/modified sources or new models when fine tuning then turn them off.

eshifri, 2023-01-13:
Wouldn't it be nice to allow "enable" to be controlled by a switch?
This way one can change performance of several CF depending on scenario.

lshems, 2023-01-13:
To meta for most users. Two switches to trigger the same thing. You can manage that if you want with logical switches as a trigger and have te functions on 'ON'.

pfeerick, 2023-01-14:
Agreed... multiple SF/GF can already be triggered via a switch by virtue of the switch condition. So while I understand where you're getting at, that isn't needed, and with the 'SAFE' SF coming in soon hopefully, a two-stage switch will also be possible.

gagarinlg, 2023-05-06:
Any decision if this should go into 2.9 or not?

pfeerick, 2023-05-08:
Yes, I want it, just needs rebasing ... unfortunately phil made changes to the SF page in the meantime that I couldn't resolve easily when I looked at it last.

PR #3601 (2023-12-03): feat: Add "Enable" button to all SF/GF (#3024)
Resolves #3024
Squashed rebase of #3038 for inclusion in v2.9