EdgeTX/edgetx #3030: Multiply or Divide telemetry values - particularly voltage

Dweinbach, 2023-01-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Problem: Telemetry on some models only measures total battery voltage, but modelers think in terms of per-cell voltage.

### Describe the solution you'd like

It would be nice to be able to divide a measured telemetry value like voltage by a hard coded (or set by SF or GV or Switch) number (like number of cells). Similar to what happens with RPM. in fact, it could be dedicated functionality when you select voltage as the units, that then lets you enter a constant, SF, or GV as the number of cells.
Alternatively, a more general purpose mult/div of telemetry value with the same options would work.
Finally, a 3rd possibility is to allow a mix to use a telemetry value as an input? Lots of ways to solve this problem. We just need one.

### Describe alternatives you've considered

1. Tried mixes - don't support telemetry inputs
2. Tried ratio in telemetry - couldnt figure out
3. Tried using SF to override a channel with the telemetry value so I could feed it to a mix - not allowed

### Additional context

none

lshems, 2023-01-13:
Inputs allow telemetry sources and multiplication. Division is as always multiplication by the inverse. 

So setting a scale in an input of 33% is similar to dividing by 100/33% = 3.

Can do for any number of cells.

But....

Indeed, it then becomes an input and is not listed as telemetry.

I would not complain if the possibility to scale telemetry on inputs is abandoned and shifted to the sensor setup. Makes much more sense.

Dweinbach, 2023-01-14:
> Inputs allow telemetry sources and multiplication. Division is as always multiplication by the inverse.
>
> So setting a scale in an input of 33% is similar to dividing by 100/33% = 3.
>
> Can do for any [...]

I tried creating an input with the telemetry battery voltage as input and a scale of 25%, but it always read out "25". the input was 1530 (15.3V). I assume that is because the max of an input is always 100, and it clamped the 1530 to 100 and then mult by 25%? I also tried a custom 25% rate curve, and same result.

So if someone DOES know how to do it, that's great - please tell me. But otherwise, scaling the telemetry sensor values is still needed/welcome

lshems, 2023-01-14:
The scale factor is providing 10^n factor. I'll check later if I can find a reference or make some instructions. Weekend and family business...

Dweinbach, 2023-01-15:
Thanks @lshems ! I've tried everything I can think of and can't figure it out. It seems like it would need a negative value of scale to turn 1,520 into 3.8 (or even 380), and negative values of scale are not supported. Looking forward to your guidance!

lshems, 2023-01-15:
ok. The ratio field should do it on the telemetry sensor setup. just tried in the simulator, and you can adjust the ratio and it does just what it describes: apply a ratio.

[image]

lshems, 2023-01-15:
I think there is something else wrong. Send me a PM on rcgroups. Lshems. with your whatsapp phone number. And i'll see if i can help.

Dweinbach, 2023-01-15:
Ratio is not a very good solution. I can put in a ratio of 640 to get a close approximation of the value, but off by several orders of magnitude. and because decimal cannot be a ratio or a fraction, you cannot divide a number with it.
After some more playing, I WAS able to come up with an OK but not ideal solution. I created 2 calculated telemetry values - one just creates a constant value of zero. The 2nd is my cell voltage, which is a calculated "average" type. by averaging (Battery,0,0,0), I can get 1/4 of the battery voltage. The only problem here is it can only read out to one decimal point, but I guess that is the same as with an actual cell measurement? 
Not to lose the point here - while I appreciate the workaround, there are lots of vehicles that only report overall battery voltage, and it would be nice to have a straightforward solution that could divide it by 1,3, or 4 depending on a switch setting, for example (or at least divide [and multiply] it by a constant with at least 2 decimal places of precision). I actually think improving the calculated telemetry feature to include "divide" would be the best, so you can keep the unmolested battery voltage as well as the calculated cell voltage.
Thanks for your help @lshems !

lshems, 2023-01-15:
I do agree this entire sensor scaling needs to be simplified

frankiearzu, 2024-02-16:
Dave, check 2.9.4.  We did a lot of work to fix scaling in 2.9.3-2.9.4.. there were hiding x10, /10, etc. that I think we fixed.

The value of 25.5 (previously displayed as 255 in 2.9.2 and before, no decimal )  means 100%
A value below 25.5 will scale down, and a value above will increase.

Propose an improvement to show % too, so in your case that you want to do 25% of a value, is easy to see,
Take a look at #4648, there is a compiled TX16 2.9.4 + this change is the pull request.

VancouverUmbrella, 2024-04-17:
Hi everyone,

I'd like to add a possibly related request: 

**The ability to divide one telemetry value by another e.g. [power in] / [vertical speed].**

An alternative implementation would be an operator to take the inverse of a telemetry value (which could in turn be used as a multiplier for another value).

Is this request relevant in this ticket? I think so but am not completely sure.

Kindly,
Andrew

blopa1961, 2024-04-18:
@VancouverUmbrella 
I requested a more general enhancement that could include a solution for this.
Please check #4785; maybe you can add your request there.