EdgeTX/edgetx #3035: Edge parameter change after edit and exit.

Kevltan, 2023-01-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If in edge command you edit the delay time and exit with for instance [1.5:<<] the second parameter reverts to [1.5:39.5]. You have to reselect the [<<] parameter before exiting or come back in and change back afterwards. It does work but it is strange that you have to revisit this parameter before exiting and check that it has remained as set.

### Expected Behavior

Change the parameter selected in edge and expect no change to second parameter when exiting.

### Steps To Reproduce

in 2.9.0 nighltly but expect is also in be 2.8.0

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X12

### Anything else?

14-01-23

philmoz, 2023-01-14:
This is covered in issue #3022 
It will be fixed when PR #3026 is merged.

pfeerick, 2023-01-15:
Duplicate of #3022

closed 2023-01-15 by pfeerick

Kevltan, 2023-01-15:
Have you updated this issue in the latest nightly as the same is true if it
has been?

Kevltan, 2023-01-15:
I don't think it is the same issue 😕

philmoz, 2023-01-16:
My apologies - this is not the same issue as #3022 and it should be re-opened.

Kevltan, 2023-01-16:
Ok, thanks. Will re-open👍

Kevltan, 2023-01-16:
How do I re-open?

pfeerick, 2023-01-17:
My apologies also - I missed that it was only on closing the edit window rather than changing when you changed model/power cycled. 

There should be a button down the buttom next to the comment button, or simply wait for someone to respond to the issue and re-open it. 
[image]